knurling-rs/defmt #452: Snapshot tests are failing without an error message

jonas-schievink, 2021-04-20:
Running `cargo xtask test-snapshot` locally results in this output:

```diff
🧪 qemu/snapshot
log (dev)
-0.000000 INFO Hello!
-0.000001 INFO World!
-0.000002 INFO The answer is 42
-0.000003 INFO Hello 42 42!
-0.000004 INFO Hello 256 42 false
-0.000005 INFO 🍕 slice [3, 14]
-0.000006 INFO 🍕 array [3, 14, 1]
-0.000007 INFO float like a butterfly 5.67 5.67
...
```

and so on, for every snapshot test. xtask should display the Cargo/rustc output here instead of just outputting this unhelpful diff.

Urhengulas, 2021-04-20:
@spookyvision Do you want to take this on?

spookyvision, 2021-04-20:
no time, sorry

Urhengulas, 2021-04-20:
@jonas-schievink Do you prefer only the rustc output, or both rustc and the diff?

jonas-schievink, 2021-04-20:
Just rustc output is enough, the diff is always going to be "everything deleted" if compilation fails

spookyvision, 2021-04-21:
(I'd keep the diff around for other cases though?)

Urhengulas, 2021-04-21:
> (I'd keep the diff around for other cases though?)

In which cases do you imagine it to be useful?

spookyvision, 2021-04-22:
the diffs are detailed info about "expected" vs "actual" output, so you'd use that to narrow a problem down

Urhengulas, 2021-05-10:
I will go forward with removing the diff in the first step and replace it with the rustc output. Later we can add the diff back, but probably guarded by a cli-flag.

Urhengulas, 2021-05-10:
> The diff should be displayed if the compilation succeeds, it just isn't of much use if it fails

akay

commit a723e610 (2021-05-28): Merge #476

476: `xtask`: `fn utils::run_capturing_stdout` r=Urhengulas a=Urhengulas

Fix #452
PR #476 (2021-05-28): `xtask`: `fn utils::run_capturing_stdout`
Fix #452
closed 2021-05-28 by bors[bot]