knurling-rs/defmt #459: Controlling logging more granularly than crate-level

mattico, 2021-04-22:
I have a project that has multiple processes running at once (16 RTIC tasks). To get enough context to solve an issue I often need to log very verbose things. Some of the logging is temporary but some of it I spend a lot of time refining so it remains useful when I come back to work on another problem with that module. 

I can't just have the logging always enabled. I have plenty of spare CPU and flash, but RTT gets blocked causing my real-time tasks to fail and there isn't a good way to filter log messages on the host side so it would be far too much noise. From what I can tell there's currently no way to enable/disable logging at anything less than crate granularity. It would be very onerous to break up my crate into smaller crates just for logging.

What I'm currently doing is creating a cargo feature to control verbose logging in each module and using a macro like this:

```rust
macro_rules! trace_usb {
    ($str:literal) => {
        if cfg!(feature = "trace-usb") {
            defmt::trace!($str);
        }
    };
    ($str:literal, $($arg:expr),+) => {
        if cfg!(feature = "trace-usb") {
            defmt::trace!($str, $($arg),+);
        } else {
            $( let _ = $arg; )+ // To avoid unused variable warnings
        }
    };
}
```

This works okay, but it's a lot of boilerplate and the log lines now point to the macro declaration rather than the use site. I'd prefer something like this:

```rust
defmt::log_if!(feature = "trace-usb", "Foobar {:?}", x);
// or just
defmt::log_if!("trace-usb", "Foobar");
```

https://github.com/knurling-rs/defmt/issues/255 discussed things related to this, but its focused on *not* using cargo features and is stalled on that.

I propose something easier: just use cargo features for now. Even more of them.

Urhengulas, 2021-04-26:
@mattico: Interesting idea! Do you see any drawbacks of this proposed solution?

mattico, 2021-04-26:
Not that I can see. It's a bit of API surface which may get replaced with something better, but I think defmt is still in the early days where it's acceptable to have some API churn.

That's why I opened the issue: to see if I'm missing something. 

Also to bikeshed the name. `log_if!` sounds like it could act on a runtime value like a normal `if` expression. `log_if_cfg!` sounds like it would take a regular `cfg!` syntax which it could, but that may be verbose or complicated to implement or otherwise undesirable. `log_if_feature!` is starting to be verbose but is the most clear.

Urhengulas, 2021-05-14:
Just noticed that this issue is similar to https://github.com/knurling-rs/probe-run/issues/74. `probe-run#74` is trying to enable more granularity using env variables.

japaric, 2021-06-24:
> Just noticed that this issue is similar to knurling-rs/probe-run#74. 

that probe-run issue is about filtering logs on the *host* side before printing them to the console.

mattico's macro filters (enables/disables) logs on the *target* side to avoid emitting some of them to save data bandwidth (e.g. USB bandwidth when using RTT). There's a related target side filtering issue, #255, which discusses env variables and reading them from the defmt proc macros (e.g. `defmt::info!`).

japaric, 2021-07-02:
@mattico would PR #519 but with module path support (i.e. if it had feature parity with `env_logger`) cover your use case? Or do you need finer granularity than modules? (certainly, Cargo features can be even more fine grained but they are less structured due to its greater flexibility)

mattico, 2021-07-02:
Module-level filtering would cover about 90% of my existing features, and I wouldn't be too concerned with keeping around the last few cargo features.

closed 2021-10-04 by bors[bot]