knurling-rs/defmt #460: Logging using UART

mattico, 2021-04-22:
I have a process that runs at 1kHz which then triggers other processes. Even being careful about what I choose to log, when something goes wrong other things go wrong and the RTT buffer fills up which blocks. Now the real-time processes are guaranteed to fail which prints more logs and the entire system grinds to a halt until I reset it.

I've spent a lot of time trying to make RTT faster. I've tried ST-Link V3s, BlackMagic probes, J-Links with official drivers and WinUSB. I didn't have too many issues with the official J-Link RTT viewer but that doesn't support defmt obviously. Maybe [hs-probe](https://github.com/probe-rs/hs-probe) will be fast enough but that doesn't exist yet. It's nice to be able to run everything through a single cable to the probe but it's just not fast enough right now.

UARTs are plentiful. You can write to them with a circular buffer and DMA with low overhead. You can use a $1 FTDI serial-to-usb to read them at 12MBaud. Serial drivers work okay cross platform and to a certain extent you can just read from them like a file. If you're already using USB in the device you can "just" add a CDC endpoint and point the host software at that to get your logs.

The difficulty is hardware support. Unlike RTT you can't just rely on the probe to handle it. However, serial writing isn't particularly hard and you can use one implementation with a few cfgs to handle an entire microcontroller family. I'm envisioning something like [synopsys-usb-otg](https://github.com/stm32-rs/synopsys-usb-otg) which provides a fairly portable abstraction.

I'll try to push this forward when I have the time but I'm interested in your thoughts.

jamesmunns, 2021-04-26:
For the record, I've found the best performance using a DapLink rather than any of the items you've mentioned (Specifically A/B tested using an nRF52-DK with stock J-Link firmware and flashing the CMSIS-DAP firmware to the DK), however I agree with you for both portability and performance reasons.

Wearing my hobbyist (not Knurling/Ferrous) hat, I've been planning on making an I2C defmt impl for my [sprocket boards](https://sprocket.jamesmunns.com], and my plan was to do something like:

* Use bbqueue, and let the user choose the buffer size (I only have 8K total on the stm32g0)
* Use the framed mode, so it's possible to get one message at a time (also bypassing framing issues)
* Silently hand the producer handle to defmt when you call some kind of `init()` function
* Give the user the consumer, so they can give it to their I2C/Serial/USB/RAM whatever driver, which acts as a "message stream"

I think this solution would allow for a reusable core that could be used for any transport.

BUT: I don't think it will be possible to guarantee that all messages "make it out alive". To counteract this, I plan on keeping a "dropped message" (or maybe "dropped message"?) counter, that can at least inform you that you should log less or increase the buffer size. I think this is a reasonable compromise. I don't know when I will get to it, but that's at least how I plan to do it, if no one gets there before me.

mattico, 2021-06-08:
I did finally get [a DapLink probe](https://www.aliexpress.com/item/1005002603229431.html?spm=a2g0s.9042311.0.0.103f4c4d58lQxq) and the flashing is 5X faster than the JLink, which is [probably due to probe-rs](https://github.com/probe-rs/probe-rs/issues/297). I'll need to see how fast the RTT is, but I had to decrease the SWD frequency to 20MHz so it may not be better.

I'll leave the framing/non-blocking to https://github.com/knurling-rs/defmt/issues/492, which I'm looking forward to.

I like the idea of using bbqueue as the interface to generic consumers. That seems more flexible, less fragile, and less work than trying to make different transports on top of HAL traits.

jamesmunns, 2021-12-14:
@mattico, I have now implemented this idea here: https://github.com/jamesmunns/defmt-bbq.

I've also included an example of end-to-end logging using a usb-serial interface (using the nrf52840).

mattico, 2021-12-14:
@jamesmunns I did end up making a fast lock-free RTT logger which uses separate RTT channels per priority level, so I don't have a use for defmt-bbq at the moment - but it looks really nice.

closed 2022-07-13 by Urhengulas