knurling-rs/defmt #463: Support non-embedded targets

Dirbaio, 2021-05-02:
Currently defmt only works in ELF targets, and requires a custom linker script. Is supporting non-embedded targets in the radar? 

- Targets that don't use ELF: **Windows, iOS, MacOS**
- Targets that do use ELF, but modifying the linker script is not as practical: **Linux, Android**

I think the most portable way would be to not rely en ELF/linker tricks and have the proc macros hit flat files (or a sqlite DB?) to assign indexes to format strings. 

This would be a big change, so if you don't want to do this because the only goal is supporting embedded devices, I understand :) I thought I'd open this issue to confirm.

Urhengulas, 2021-05-05:
This certainly sounds like a desirable feature and feasible if someone likes to contribute or fund it.

DBLouis, 2021-08-05:
Is it possible to make defmt works on standard x86 linux? It fails to link with the custom linker script. If I don't use the custom linker script `readelf` still shows a section that contains my log strings. However when running `defmt-print` on the ELF it fails with: `Error: .defmt data not found`.

japaric, 2021-08-05:
> Is it possible to make defmt works on standard x86 linux?

No OS is supported at the moment; that's what this issue is about. What you are describing are the symptoms of several unimplemented things.

> It fails to link with the custom linker script.

sounds about right. wouldn't know the exact reason off the top of my head but may be related to Linux binaries being relocatable. Firmware images use absolute addresses for everything (relocation=static in linker/compiler lingo). This ("format string 'interning'" is probably the hardest thing to fix / design / implement. More so on platforms that don't use the ELF format: macOS and Windows.

> If I don't use the custom linker script readelf still shows a section that contains my log strings.

(interesting that they "survive" (are not discarded by the linker) the linking process)

> However when running defmt-print on the ELF it fails with: Error: .defmt data not found.

without the custom linker script the log strings won't end in the right place so the .defmt linker section is not created hence the error message

---

The other thing that's missing is a transport crate for Linux and other OSes. That crate defines *how* the encoded/compressed defmt data is output (e.g. framing) and to where (e.g. stdout, a file, a named pipe, a socket, etc.). You would then pass this defmt data to `defmt-print`

DBLouis, 2021-08-05:
Is there any hope of making it work by setting the right compiler flags so that the log strings are "at the right place"? I don't know much about linking so maybe I am missing the point

japaric, 2021-08-05:
maybe? I'm familiar with the details of *static* linking but not the details of dynamic / relocatable linking so I can't answer that

DBLouis, 2021-08-05:
With the following parameters linking succeed but the program segfault immediately.
I think the default `defmt` script might overwrite stuff.

```
  -C relocation-model=static  -C target-feature=+crt-static  -C link-arg=-Tdefmt.x
```

DBLouis, 2021-08-07:
I had assume that the default linker script was always passed as a parameter. Looks like is it not. After adding it to `.cargo/config` my program compiles and runs, and `defmt-print` decode frames correctly.

```sh
[target.x86_64-unknown-linux-musl]
rustflags = [
  "-C", "target-feature=+crt-static",
  "-C", "relocation-model=static",
  "-C", "link-arg=-T/usr/x86_64-linux-musl/usr/lib/ldscripts/elf_x86_64.x",
  "-C", "link-arg=-Tdefmt.x"
]

[build]
target = "x86_64-unknown-linux-musl"
```

Urhengulas, 2021-08-19:
> It seems to works fine without `+crt-static` but not without `relocation-model=static`. For both musl and glibc.

@DBLouis, very cool that you made it work! 🎉  What is your usecase that makes you want to use `defmt` over `log`?

andresv, 2021-11-07:
Use case can be that there is a `no_std` lib that is primarily meant for embedded devices and it has bunch of `defmt` prints in there and you wanna use this lib also on x86.

It seems one way to do it is just to add different feature flags to the lib itself: https://github.com/embassy-rs/embassy/blob/master/embassy/src/fmt.rs

newAM, 2021-11-07:
> Use case can be that there is a `no_std` lib that is primarily meant for embedded devices and it has bunch of `defmt` prints in there and you wanna use this lib also on x86.

That would work, but I think the goal is to have binary logging on hosted targets, for things like embedded linux where there are storage limitations with normal logging.

gauteh, 2022-06-26:
Came across this while working on a defmt-serial target: https://github.com/gauteh/defmt-serial. Wrote an example for the host environment, works with: https://github.com/knurling-rs/defmt/issues/463#issuecomment-894677833:

```
❯ DEFMT_LOG=debug cargo run
   Compiling defmt-serial v0.1.0 (/home/gauteh/dev/embedded/defmt-serial)
   Compiling example-std v0.1.0 (/home/gauteh/dev/embedded/defmt-serial/example-std)
    Finished dev [unoptimized + debuginfo] target(s) in 0.54s
     Running `bash defmt-print-runner.sh target/x86_64-unknown-linux-musl/debug/example-std`
Hello, world!
Logging to info with defmt..
Good bye.
INFO  Hello defmt-world!
└─ example_std::main @ src/main.rs:28
DEBUG Now at: 0
└─ example_std::main @ src/main.rs:31
DEBUG Now at: 1
└─ example_std::main @ src/main.rs:31
DEBUG Now at: 2
└─ example_std::main @ src/main.rs:31
DEBUG Now at: 3
└─ example_std::main @ src/main.rs:31
DEBUG Now at: 4
└─ example_std::main @ src/main.rs:31
DEBUG Now at: 5
└─ example_std::main @ src/main.rs:31
DEBUG Now at: 6
└─ example_std::main @ src/main.rs:31
DEBUG Now at: 7
└─ example_std::main @ src/main.rs:31
DEBUG Now at: 8
└─ example_std::main @ src/main.rs:31
...
DEBUG Now at: 45
└─ example_std::main @ src/main.rs:31
DEBUG Now at: 46
└─ example_std::main @ src/main.rs:31
DEBUG Now at: 47
└─ example_std::main @ src/main.rs:31
DEBUG Now at: 48
└─ example_std::main @ src/main.rs:31
DEBUG Now at: 49
└─ example_std::main @ src/main.rs:31
WARN  Done!
└─ example_std::main @ src/main.rs:34
^C
defmt-serial/example-std on  main via 🦀 v1.61.0-nightly
```

Urhengulas, 2022-06-27:
That is great to see @gauteh! Is there anything we can do to support non-embedded targets better?

silvergasp, 2022-07-07:
If anyone is working on this, you might find the linker scripts and tooling in the [google/pigweed/pw_tokeniser](https://github.com/google/pigweed/tree/main/pw_tokenizer) directory useful. It's all written in c++ but works for host/embedded-arm but uses similar mechanisms to what defmt uses. i.e. pw_tokeniser and defmt offer very similar functionality.

elpiel, 2023-08-01:
Hello,
I saw this issue and since I don't see much work around it after 2 years I was wondering if there's a way to circumvent defmt logging all together?

Right now https://github.com/knurling-rs/defmt/issues/463#issuecomment-894677833 seems to be working for linux `-gnu` as well, the only problem I'm encountering is that doc tests started failing.
I'm using `defmt_serial` with an `StdoutSerial` impl from the `std` example.

```
note: /usr/bin/ld: /mnt/Disk/Projects/lechev.space/netbox/netbox-hardware/target/x86_64-unknown-linux-gnu/debug/deps/libnetbox_firmware-5b6d3b1c9ed15201.rlib(netbox_firmware-5b6d3b1c9ed15201.14bxe1s9urqapwuu.rcgu.o): relocation R_X86_64_32 against symbol `rust_eh_personality' can not be used when making a PIE object; recompile with -fPIE
          /usr/bin/ld: failed to set dynamic section sizes: bad value
          collect2: error: ld returned 1 exit status
```

Urhengulas, 2024-12-02:
Closed in favor of https://github.com/knurling-rs/defmt/issues/463.

jonathanpallant, 2024-12-02:
Ah, that is an interesting distinction. Thank you for clarifying.

Just to check I understand this:

1. defmt currently writes interned strings into symbol names, and the symbols are placed by the linker into the `.defmt (NOLOAD)` section.
2. There is a need for this mechanism to basically work as-is on platforms like Windows/macOS/Linux/Android/iOS, where defmt-decoder needs to work out how to get the interned strings out of the PE, Mach-O or ELF executable.
3. There is an additional, but separate, need, for the strings to optionally be available at run-time (instead of being interned) so as to enable run-time filtering and/or formatting. That requires (2), but also additional work. It is important this is optional.

I'm going to re-open this and rename the issue so we keep track of it.

Dirbaio, 2024-12-02:
getting "json in symbol names" working on all OSs might be challenging. Maybe it's easier to find another way to do interning that works independently of the OS or executable format. For example the proc macro could access some file at compile time (sqlite maybe?) to do it.

jonathanpallant, 2024-12-02:
So, trying this with a basic `defmt-file` implementation, making the log file is easy. But defmt-print can't handle a Linux ELF file that was created without using the `defmt.x` linker script to move stuff into appropriately named sections. So, we probably need to teach `defmt-decoder` to be a bit smarter, because the symbols are there:

```console
root@d9db5f417f28:/work# objdump -Ct ./target/debug/defmt-demo  | grep '\.defmt'
0000000000048b7d l    d  .defmt.{"package":"defmt-demo","tag":"defmt_write","data":"{=u64}","disambiguator":"14958850188590970431","crate_name":"defmt_demo"}	0000000000000000              .defmt.{"package":"defmt-demo","tag":"defmt_write","data":"{=u64}","disambiguator":"14958850188590970431","crate_name":"defmt_demo"}
0000000000048b7e l    d  .defmt.{"package":"defmt-demo","tag":"defmt_println","data":"Hello, world!","disambiguator":"16717065360975528047","crate_name":"defmt_demo"}	0000000000000000              .defmt.{"package":"defmt-demo","tag":"defmt_println","data":"Hello, world!","disambiguator":"16717065360975528047","crate_name":"defmt_demo"}
0000000000048b7d g     O .defmt.{"package":"defmt-demo","tag":"defmt_write","data":"{=u64}","disambiguator":"14958850188590970431","crate_name":"defmt_demo"}	0000000000000001              {"package":"defmt-demo","tag":"defmt_write","data":"{=u64}","disambiguator":"14958850188590970431","crate_name":"defmt_demo"}
0000000000048b7e g     O .defmt.{"package":"defmt-demo","tag":"defmt_println","data":"Hello, world!","disambiguator":"16717065360975528047","crate_name":"defmt_demo"}	0000000000000001              {"package":"defmt-demo","tag":"defmt_println","data":"Hello, world!","disambiguator":"16717065360975528047","crate_name":"defmt_demo"}
```

On the subject of JSON symbol names, yeah, some platforms don't like that. Rather than hex encode them I was hoping to use a transformation that doesn't make any ASCII text unreadable.

Interning into something other than a magically named symbol leaves you with the problem of coming up with a unique string ID. I suppose the proc-macro could write to a file as a side-effect of expanding a macro call, but that seems like it would need some investigation.

jonathanpallant, 2024-12-02:
Work in progress: https://github.com/knurling-rs/defmt-experiments

jonathanpallant, 2024-12-02:
macOS symbols:

```console
$ objdump -t ./target/debug/defmt-file-demo | rg '\.defmt'
0000000000000000      d  *UND* /Users/jonathan/Documents/ferrous-systems/defmt-experiments/defmt-file-demo/target/debug/deps/libdefmt-2d61f47589874c51.rlib(defmt-2d61f47589874c51.defmt.90156f26d6483a20-cgu.0.rcgu.o)
0000000100054003 g     O .defmt,end.ENCODING __defmt_encoding_ = rzcobs
0000000100054002 g     O .defmt,end.VERSION __defmt_version_ = 4
0000000100054001 g     O .defmt,49d3406bedb6ffea _{"package":"defmt-file","tag":"defmt_write","data":"{=u64}","disambiguator":"16360403663171214326","crate_name":"defmt_file"}
0000000100054000 g     O .defmt,8addfb286f1161b6 _{"package":"defmt-file-demo","tag":"defmt_println","data":"Hello, world!","disambiguator":"13817463997336866179","crate_name":"defmt_file_demo"}
```

Dirbaio, 2024-12-02:
For many use cases it's not a must that defmt can embed the table in the executable itself. If the macros use a file to intern, the decoder could just read that file (maybe with postprocessing to convert it to the [defmt-json-schema](https://github.com/knurling-rs/defmt/tree/main/decoder/defmt-json-schema) format?). This way there's no need to embed symbols with cursed names.

Dirbaio, 2024-12-02:
I think it should be doable. 

Option A: macro produces `defmt-json-schema` directly. I'm not sure if this is doable with good perf.
Option B: macro produces some "raw" file in a format that's more suitable for modifying. It could be sqlite, it could be some lighter key-value storage format.

The macro would:
- Lock the file so it doesn't race with other macro invocations
- Open the file (or create if it doesn't exist)
- Search whether there's already an ID with the exact same JSON. If so, use that one.
- If not, pick the next free ID, add it to the file.
- Close and Unlock the file.

The "search" part would have to be `O(n)` if the macro produces `defmt-json-schema` directly. It could be faster with sqlite or some k-v store if we use the JSON as *key* and the ID as *value* (ie the reverse of what you'd normally do!)

## User experience

The user has two "interner backends" to choose from.

- `DEFMT_INTERNER=elf` -> The output is an ELF file with an embedded table.
- `DEFMT_INTERNER=file` -> The output is an executable (of whatever format, of any OS) and a "raw" defmt table file.

then we'd have an utility that can:

- convert from "raw table" to `defmt-json-schema`.
- convert from "executable with embedded table" to `defmt-json-schema`.
- Decode, with inputs "binary log stream" and "raw table"
- Decode, with inputs "binary log stream" and "executable with embedded table"
- Decode, with inputs "binary log stream" and `defmt-json-schema`

> Like, who truncates the file at the start of a compilation?

The macro can't do it safely, the user would have to do it. We'd have to document when and how should you do it. 

Something interesting is if the user doesn't delete the file, they'd get two builds that actually decode correctly with the same table. This might actually be a *feature*, not a bug. You could build e.g. a bootloader and an app sharing a table, or N builds for multiple hardware revisions sharing a table so you can decode them all without caring about the revision. It might be a better solution for #652. 

This'll all need documenting, of course...

> What if half your crates have already been compiled and the other half are being compiled? How do you still get a unique 16-bit ID for each interned string?

If you build the same crate again, the macro would look up the JSON and find it already has an ID and reuse that. If you edit the crate and the text or line numbers or whatever change it'd get assigned a new ID, leaving the previous ID unused.

As you edit+run things you'd run out of 16bit IDs, then you'd have to delete the file and rebuild from scratch. (maybe we should investigate how to change IDs to be 32bit...)

For "production" builds this wouldn't be a problem if the user ensures to build them in CI with a clean table file and not from a random developer laptop...

jonathanpallant, 2024-12-02:
I like it. I think it would have to be a defmt feature, so as to force recompilation when it changes (ugh, imagine half the strings being in the ELF and half being in the database), but I can imagine it working as you have outlined.

Are you able to work on a proof-of-principle, or is that something we should try and resource?