knurling-rs/defmt #491: bug in zero padding display hint

Urhengulas, 2021-05-23:
This code:
```rust
defmt::info!("{:00x}", 42);
defmt::info!("{:01x}", 42);
defmt::info!("{:02x}", 42);
defmt::info!("{:03x}", 42);
defmt::info!("{:04x}", 42);
defmt::info!("{:05x}", 42);
defmt::info!("{:06x}", 42);
```
results in these logs:
```
0 INFO  0x2a
└─ hello::__cortex_m_rt_main @ src/bin/hello.rs:8
1 INFO  0x2a
└─ hello::__cortex_m_rt_main @ src/bin/hello.rs:9
2 INFO  0x2a
└─ hello::__cortex_m_rt_main @ src/bin/hello.rs:10
3 INFO  0x2a
└─ hello::__cortex_m_rt_main @ src/bin/hello.rs:11
4 INFO  0x2a
└─ hello::__cortex_m_rt_main @ src/bin/hello.rs:12
5 INFO  0x02a
└─ hello::__cortex_m_rt_main @ src/bin/hello.rs:13
6 INFO  0x002a
└─ hello::__cortex_m_rt_main @ src/bin/hello.rs:14
```
---
I'd expect that a display hint `:04x` results in `0x002a` for `42`, but it currently seems to include the `0x` in the length and only results in `0x2a`. `06x` will currently result in the desired `0x002a`, but I would expect `0x00002a`.

Urhengulas, 2021-05-23:
@derekdreery Would you like to look into this?

jonas-schievink, 2021-05-23:
This matches the behavior of `core::fmt`, so it might be better to keep it

Urhengulas, 2021-05-23:
Really? that is weird, but okay with me 🤔

jonas-schievink, 2021-05-23:
Hmm, actually our output for `{:06x}` matches `{:#06x}` with `core::fmt` - it doesn't include the `0x` unless you use `#`, so maybe it's fine if we change how the padding works too?

Urhengulas, 2021-05-23:
> Hmm, actually our output for `{:06x}` matches `{:#06x}` with `core::fmt` - it doesn't include the `0x` unless you use `#`, so maybe it's fine if we change how the padding works too?

That is also what I just tested and was about to write 😅 

I think consistency is helpful to help people not getting confused, about display hints. I'd rather vote for matching the behavior, and explude the `0x`.

derekdreery, 2021-05-24:
When I made this, I tried to merge the beahvior of core::fmt and the
existing defmt formatting. Happy to do a PR to change it if that's the
consensus.

Urhengulas, 2021-05-27:
@derekdreery: I would be happy if you go forward to adapt the output to match `core::fmt`! 😄

derekdreery, 2021-06-01:
Hi, so there are a few differences between defmt and core::fmt:

 1. In core::fmt you have `:x` and `:X` which will use lower and upper case a-f respectively.
 2. In core::fmt, the alternative `:#x` rather than `:x` includes the `0x` at the beginning, in defmt `0x` is always included.

The might be others. Would you like me to change behaviour to match for 1 and 2, just 1, or something else?

Urhengulas, 2021-06-06:
Number one is already the case for `defmt`, isn't it? Thinking about it again I think including the `0x` isn't that bad of an idea, therefore I think no action is required.

derekdreery, 2021-06-06:
@Urhengulas are you sure that having `:x` and `:X` differ to `core::fmt` is good? I've currently got a MAC address I'd like to format as `AA:BB:11:22:33:44` but currently can't do this because I get `0xAA:0xBB:0x11:0x22:0x33:0x44`. In `core::fmt` you can still get the `0x` prefix by using the alternative formatter (`:#x` and `:#X`).

Urhengulas, 2021-06-06:
I see. I mainly thought about logging single hex numbers, but for something like a mac-address it makes sense to be able to exclude the `0x`. I then would agrre, that completely matching the `core::fmt` behavior makes sense 😄

commit 19f92c59 (2021-06-11): Merge #503

503: Add alternate hint ('#') r=derekdreery a=derekdreery

Also make formatting match `core::fmt`.

This changes the current behavior, so some thought should be given before accepting. I expect it will break some tests: I'm not sure how to run them all locally so I will rely on CI to find them.

Closes #491
PR #503 (2021-06-11): Add alternate hint ('#')
Also make formatting match `core::fmt`.

This changes the current behavior, so some thought should be given before accepting. I expect it will break some tests: I'm not sure how to run them all locally so I will rely on CI to find them.

Closes #491
closed 2021-06-11 by bors[bot]