knurling-rs/defmt #494: semver-breaking: "Unused variable" errors with deny(unused).

Dirbaio, 2021-05-28:
The following used to build fine with defmt 0.2.1, and now fails with defmt 0.2.2. This is due to #477 

```rust
#![deny(unused)]
let x = 1; // error: unused variable: `x` if no "defmt-xxx" features are enabled
info!("x={}", x);
```

jonas-schievink, 2021-05-28:
Also ran into this while working with a client today

Urhengulas, 2021-05-29:
Options are reverting or releasing a new minor version?

jonas-schievink, 2021-05-29:
We should change how the macro expands to silence the warning, then publish a new patch release

Urhengulas, 2021-05-30:
My idea would be to just match the `args`. This should look sth like following but I am struggeling with the macro syntax:
```rust
cfg_if::cfg_if! {
    if #[#logging_enabled] {
        ...
    } else {
        match (#(&(#args)),*) {
            _ => {}
        }
    }
}
```
Currently I am getting a `error[E0433]: failed to resolve: use of undeclared crate or module 'cfg_if'`, although I specified it as a dependency for `defmt-macros`.

Urhengulas, 2021-05-30:
Turns out I also need to add it to `defmt`. I guess it makes sense 🤔

Urhengulas, 2021-05-30:
And it also turns out I don't need to add it to `defmt-macros`. That is a bit weird to me 😕

closed 2021-05-31 by bors[bot]

Urhengulas, 2021-05-31:
@jonas-schievink I would release a new version. Should we yank the current one?

jonas-schievink, 2021-05-31:
Yeah, probably

commit 4ca63c09 (2021-05-31): Merge #497

497: `macros`: match unused vars if logging is disabled r=jonas-schievink a=Urhengulas

Fixes #494
PR #497 (2021-05-31): `macros`: match unused vars if logging is disabled
Fixes #494