knurling-rs/defmt #506: Backwards-compatability tests

Urhengulas, 2021-06-11:
#503 broke the backwards-compatability tests and since we don't guarantee backwards-compatability we kinda disabled them by just running with the current decoder. 

My question is now if we should just remove these tests all together since we don't guarantee it anyways, or if we should pin it to the current version of the decoder?
I am fine with both with a slight preference for removing.

Urhengulas, 2021-06-12:
While thinking about it again I thought that it is already valuable to know how backwards-comaptible our output is. So keeping the tests around gives us this certainty if someone ever comes around and wants to know this.

---

@jonas-schievink what are your thoughts?

jonas-schievink, 2021-06-12:
The purpose of those tests was to make sure that the earliest semver-compatible defmt release can still decode what the current git version produces. The result here does *not* necessarily have to match 1-to-1 (so the changes to the output were fine), but I can see that that does fail the tests at the moment. Not totally sure how to do this better, maybe we just have to check that the decoder didn't fail?

Urhengulas, 2021-06-14:
So I think it would work if we just always install `probe-run` from `main` and check that the decoding doesn't fail? Thereby we make sure we don't loose decoding functionality, but enable PR authors to alter the output.

japaric, 2021-07-16:
I think we'll need to revise those backcompat tests after #287 is implemented because after that the behavior is going to be: 

breaking changes in the encoding / decoding are allowed in patch versions (but breaking API changes are not); using defmt 0.3.x anywhere in the dep graph = you use the latest patch version of defmt = no breakage in code because of the 'no breaking API changes'; probe-run will error (or maybe it will work but not print any logs?) and tell you to upgrade it if it sees a defmt patch version newer than what it supports.

so using the v0.3.x encoder with the 0.3.0 decoder (which the backcompat tests tested) is not something that's going to be part of  the workflow anymore

Urhengulas, 2021-07-20:
After #287 we should additionally to the output-format also check backwards-incompatible changes to (quoted from the PR): "the symbol naming scheme, the symbol and section layout, the data encoding / wire format".

japaric, 2021-09-10:
I think what we need to do here to test for changes that break backwards compatibility (after #540) is, at a high level:

- add a new xtask, say `test-backcompat`
- this xtask will run the same set of firmware examples as `test-snapshot` BUT
  - it will use an "old version" of `qemu-run`
  - it will *not* check the output (stdout/stderr) of `qemu-run`
  - it will check the exit code of `qemu-run`; it should be 0 (success) in all cases
  - if `qemu-run` fails, the xtask must indicate *which* test failed and print whatever `qemu-run` printed -- so we can figure out where decoding failed

By "old version" of `qemu-run` I mean a version that uses the oldest git commit of `defmt` that last bumped [this version number](https://github.com/knurling-rs/defmt/blob/59d20748785e14739a7460a468dade45579caf28/src/lib.rs#L32).

how to *install* an "old version" of `qemu-run`? The *easiest* way is to run `cargo install --git https://github.com/knurling-rs/defmt --rev $git_hash` but this is *not* great for running CI locally because it will override any local `qemu-run` (though it's unlikely people have that tool installed). An alternative that does not do that would be nice but not a deal breaker.

how to *use* an "old version" of `qemu-run`? in the `firmware/qemu` directory, `cargo run` is configured (see `.cargo/config.toml`) to use the local, i.e. latest version, of `qemu-run`. To override this one can set the `CARGO_TARGET_${target_triple}_RUNNER` env var. Like this:
``` console
$ CARGO_TARGET_THUMBV7M_NONE_EABIHF_RUNNER=path/to/binary cargo run --bin hello
```
Note that you need to specify the target triple in UPPERCASE letters.

What do we when this `test-backcompat` test breaks in a pull request?
- temporarily disable the `test-backcompat` test to land the PR
- in a follow-up defmt PR
  - bump the defmt version number in `src/lib.rs`
  - update `xtask` to use the commit hash of the pull request merge that broke backcompat

commit 37462411 (2021-10-01): Merge #592

592: xtask: add backward compability test r=Urhengulas a=japaric

**UPDATE**: I have left the defmt-version as it is since we have not made a defmt 0.3.0 release. I have also tweaked the defmt-test to not exit with non-zero exit code and remove some test-specific logic from xtask. See individual commits messages for details.

---

implements https://github.com/knurling-rs/defmt/issues/506#issuecomment-916869244
closes #506

I tested this against c7b20d53861d0c4fa476db4791d5745412f4de57 (I called that `defmt-version-3` locally) and found that a bitflags change broke the wire format. [... cut]
commit 3e2b6012 (2021-10-04): Merge #592

592: xtask: add backward compability test r=Urhengulas a=japaric

**UPDATE**: I have left the defmt-version as it is since we have not made a defmt 0.3.0 release. I have also tweaked the defmt-test to not exit with non-zero exit code and remove some test-specific logic from xtask. See individual commits messages for details.

---

implements https://github.com/knurling-rs/defmt/issues/506#issuecomment-916869244
closes #506

I tested this against c7b20d53861d0c4fa476db4791d5745412f4de57 (I called that `defmt-version-3` locally) and found that a bitflags change broke the wire format. [... cut]
PR #592 (2021-10-04): xtask: add backward compability test
**UPDATE**: I have left the defmt-version as it is since we have not made a defmt 0.3.0 release. I have also tweaked the defmt-test to not exit with non-zero exit code and remove some test-specific logic from xtask. See individual commits messages for details.

---

implements https://github.com/knurling-rs/defmt/issues/506#issuecomment-916869244
closes #506

I tested this against c7b20d53861d0c4fa476db4791d5745412f4de57 (I called that `defmt-version-3` locally) and found that a bitflags change broke the wire format. Not a big deal but I guess we should bum [... cut]
closed 2021-10-04 by bors[bot]