knurling-rs/defmt #515: defmt_rtt: flush API

japaric, 2021-06-21:
# What

`defmt_rtt::flush()` is a blocking API that waits until the host has read all the pending data in the target's (caller's) RTT buffer

# Use case

Avoid decoding errors when doing a software reset, or calling `cortex_m::asm::{bootstrap,bootload}`

For example, this likely produces a decoding error in `probe-run`

``` rust
defmt::info!("about to reset!");
SCB::sys_reset();
```

To work around the decoding error you can add a delay

``` rust
defmt::info!("about to reset!");
cortex_m::asm::delay(1_000_000);
SCB::sys_reset();
```

It's not obvious how large to make the delay so the `flush` API would be the preferred option

``` rust
defmt::info!("about to reset!");
defmt_rtt::flush();
SCB::sys_reset();
```

# Unresolved questions

- Can we put the API in the `defmt` crate by making it transport agnostic somehow?
  - the problem with a transport specific API is that it cannot be used in libraries unless one is OK with forcing all users of the libraries to use that specific transport
- in the `defmt_rtt` implementation, how do we prevent the issue of blocking forever if no host / debugger is attached?

Dirbaio, 2021-07-17:
This could be a `defmt::flush()` function that simply calls `flush()` on the global Logger. Impls where "flush" doesn't make sense can define it to do nothing, it could even have a default impl in the trait that does nothing.

With rzCOBS this is less critical though: you can still lose frames, but thanks to framing the decoder is able to recover.

japaric, 2021-07-19:
> This could be a defmt::flush() function that simply calls flush() on the global Logger. Impls where "flush" doesn't make sense can define it to do nothing, it could even have a default impl in the trait that does nothing.

in that case, this sounds backwards compatible to add. we could add a `flush` method that has a default implementation of "do nothing" to the `Logger` trait

japaric, 2021-07-22:
Implementation steps off the top of my head:

- add a new `flush` method to the `Logger` trait
  - because 0.3.0 is not out yet this can be a "required method" (= does *not* have a default implementation)

- the above will cause existing `Logger` implementations to break e.g. `defmt-rtt`; that will need to be fixed
  - the `flush` implementation for RTT transport is going to be (I think) busy wait until the "read" pointer (advanced by the host) has caught up the "write" pointer (advanced by the target)
  - that naive RTT `flush` implementation WILL block forever if no host is connected ("read" pointer never changes) = a bad thing. unsure how to best solve that; as a starter we could do:
    - if `read == write` then immediately exit `flush`
    - check the read pointer (before any loop); if it's `0` then probably no host is connected (?) so exit `flush`
    - otherwise go into the `while current_read() != current_write()` loop
    - every "few iterations" (this is the heuristics) check if the "read" pointer has changed at all (compared to the last check done every "few iterations"); if it hasn't then the host is probably not connect so exit `flush`

- add `defmt::flush`. what this should do under the hood is (I think):
  - call `Logger::acquire`  -- NOTE here by `Logger` I meant the single `Logger` implementation in the dependency graph, not the trait (you cannot call `Trait::selfless_static_method`)
  - call `Logger::flush`
  - call `Logger::release`

Urhengulas, 2021-07-22:
Excerpt from internal chat:

* How to implement `defmt-semihosting`?
  > In QEMU those should auto-flush on each write call because the target is emulated. With hardware, the write calls block for several hundred of milliseconds so that should be enough to flush the data. 'do nothing' seems fine in either case.

* How to implement defmt-itm?
  > I think that one should busy wait on the BUSY (?) flag being cleared (?). the BUSY flag is also checked in the "write" operation of the logger; you should be able to copy the logic that does the check from there into flush

commit 59d20748 (2021-09-08): Merge #550

550: `defmt::flush()` r=jonas-schievink a=Urhengulas

Fixes #515 

## TODO
- [x] skeleton: add method to trait, TODO-implements in crates and tests
- [x] defmt-rtt
- [x] defmt-semihosting
- [x] defmt-itm
- [x] adapt delays based on docs and testing
PR #550 (2021-09-08): `defmt::flush()`
Fixes #515 

## TODO
- [x] skeleton: add method to trait, TODO-implements in crates and tests
- [x] defmt-rtt
- [x] defmt-semihosting
- [x] defmt-itm
- [x] adapt delays based on docs and testing
closed 2021-09-08 by bors[bot]