knurling-rs/defmt #517: Obtain git hash from `git_version` crate

Urhengulas, 2021-06-23:
Use the `git_version` crate for obtaining the git hash, instead of the custom build-script we used so far. Similar to https://github.com/knurling-rs/probe-run/pull/221

Urhengulas, 2021-06-23:
@japaric We discussed the possibility to completely get rid of `build.rs` with this change. But since we still want to alter the linker-script and insert the version, this isn't really possible is it?

japaric, 2021-06-24:
@Urhengulas correct, we need the build script to put the `defmt.x` linker script in the project's `target` directory otherwise the linker won't find it. So we want to keep this line even if we don't modify `defmt.x` in the build script.
https://github.com/knurling-rs/defmt/blob/fde2aab0bc541081d7def42fb049a36fa793a2bc/build.rs#L38

Urhengulas, 2021-06-24:
> @Urhengulas correct, we need the build script to put the `defmt.x` linker script in the project's `target` directory otherwise the linker won't find it. So we want to keep this line even if we don't [...]

But we want to modify it, don't we? To insert the version into it?

japaric, 2021-06-24:
> But we want to modify it, don't we? To insert the version into it?

yes, we want to continue doing that. (the last line of my last comment was referring to an hypothetical scenario)

commit 7d5aaedd (2021-07-19): Merge #518

518: `build.rs`: Obtain version from macro; simplify r=Urhengulas a=Urhengulas

Fixes #517.
PR #518 (2021-07-19): `build.rs`: Obtain version from macro; simplify
Fixes #517.
closed 2021-07-19 by bors[bot]