knurling-rs/defmt #530: write a 0.2.x -> 0.3.0 migration guide

japaric, 2021-07-15:
0.3 will include several breaking changes. this guide would explain how to migrate code that uses defmt 0.2.x to 0.3.0

example of a migration guide: https://rtic.rs/0.5/book/en/migration.html

Urhengulas, 2021-07-21:
Should this guide only list breaking changes, or also new features?

japaric, 2021-07-21:
the focus should be on *fixing* things that no longer compile or no longer behave the same: changes in syntax; changes in API; `defmt-*` Cargo features no longer being needed; `DEFMT_LOG` now being a thing. It should offer a quick fix (e.g. to get the old behavior use `DEFMT_LOG=info cargo r`), ideally in the form of a `diff` when code is involved; the migration guide can defer / link to other dedicated sections of the book for details about new features (e.g. ISO timestamp display hint).

commit da1682ff (2021-10-19): Merge #547

547: Migration guide `v0.2.x` to `v0.3.0` r=japaric a=Urhengulas

Migration guide from `defmt v0.2.x` to version `v0.3.0`.

https://deploy-preview-547--admiring-dubinsky-56dff5.netlify.app/migration-02-03.html

Fixes #530.

### TODO
- [x] #505: Logger trait v2
- [x] #521: [3/n] Remove u24
- [x] #522: Replace `µs` hint with `us`
- [x] (no adaption needed) ~~#508: [5/n] Format trait v2~~
- [x] #519: `DEFMT_LOG`
- [x] #550: `defmt::flush()`
- [x] https://github.com/knurling-rs/probe-run/issues/198, https://github.com/knurling-rs/probe-run/pull/250,
closed 2021-10-19 by bors[bot]