knurling-rs/defmt #536: format string parser error should point to the docs

japaric, 2021-07-16:
For example:
``` console
ERROR: Format string error detected.
HELP: `defmt` uses a slightly different syntax than regular formatting in Rust.
  See https://defmt.ferrous-systems.com/macros.html for more details.
```

jonathanpallant, 2023-03-08:
Current you see:

```rust
defmt::println!("Hello, world!, p1 = {:z}", p);
```

```text
error: unknown display hint: "z"
  --> src/bin/hello.rs:12:21
   |
12 |     defmt::println!("Hello, world!, p1 = {:z}", p);
   |                     ^^^^^^^^^^^^^^^^^^^^^^^^^^

error: could not compile `test-pointers` due to previous error
```

jonathanpallant, 2023-03-08:
I have something that now shows:

```text
error: unknown display hint: `"z"`

         = help: `defmt` uses a slightly different syntax than regular formatting in Rust. See https://defmt.ferrous-systems.com/macros.html for more details.

  --> src/bin/hello.rs:12:21
   |
12 |     defmt::println!("Hello, world!, p1 = {:z}", p);
   |                     ^^^^^^^^^^^^^^^^^^^^^^^^^^

error: could not compile `test-pointers` due to previous error
```

commit 9d2bfde1 (2023-03-10): Uses the new Error enum to print help text.

Adds help text for Error::UnknownDisplayHint.

Closes #536
commit d4092090 (2023-03-10): Merge #735

735: Fix #536 - add help for format string parsing error r=Urhengulas a=jonathanpallant

1) Replaces `Cow<str>` with an `enum Error` to accurately describe the parse error
2) Uses `thiserror` to convert the `Error` enum into strings, which match the strings previously returned from the `parse` function
3) Uses that enum to add help text for Error::UnknownDisplayHint.

Closes #536
PR #735 (2023-03-10): Fix #536 - add help for format string parsing error
1) Replaces `Cow<str>` with an `enum Error` to accurately describe the parse error
2) Uses `thiserror` to convert the `Error` enum into strings, which match the strings previously returned from the `parse` function
3) Uses that enum to add help text for Error::UnknownDisplayHint.

Closes #536
closed 2023-03-10 by bors[bot]