knurling-rs/defmt #541: [RFC] `defmt::println!`

japaric, 2021-07-19:
## Summary

add an equivalent of `std::println!` to `defmt`

## Rationale

`defmt` log macros emulate the `log` crate; log statements have log levels (ERROR, WARN, INFO, DEBUG, TRACE) and can be filtered out (Cargo feature (current mechanism) or DEFMT_LOG (proposed in PR #519)).

A `defmt::println!` statement would have no log level associated to it and it would not be possible to filter it out.

`defmt::println` could be used in `defmt-test` to ensure that the test suite progress messages are always printed regardless of the log filter configuration -- this would make the behavior of `defmt-test` closer to the "standard" `#[test]` harness.

## Alternatives

Not doing this means that if we approve PR #519 we need some to ensure that test suite progress messages are printed even when `DEFMT_LOG` is not set.
This could mean either:

- maintaining the existing Cargo feature based filter mechanism alongside the `DEFMT_LOG` filter mechanism (this would entail figuring out how the 2 should interoperate and extra maintenance work compared to removing the Cargo feature based filter)
- OR figuring out some other mechanism

japaric, 2021-07-29:
@jonas-schievink thoughts on this? do you think it would address the `defmt-test` issue that PR #519 introduces? any other issue that #519 creates that this doesn't address?

jonas-schievink, 2021-07-29:
I don't think I really have too strong of an opinion on this at the moment. A `println!` equivalent does sound useful just by itself, and if it lets us resolve the `defmt-test` issues then that's great, on the other hand I don't think this will end up being less work than supporting the Cargo feature based filters in addition to `DEFMT_LOG`.

japaric, 2021-07-29:
Maintenance cost aside (in my mind maintaining 2 features is more work than maintaining 1 (+)) I think that for end users is going to be confusing to have 2 ways to do the same thing: "which one do I pick? when do I use one over the other?"; those questions would need to be answered with documentation (*and* when need to specify the interaction between the two features).

If I had to pick one of the 2 filter mechanisms I would go with `DEFMT_LOG` because is closer to `RUST_LOG` which is the de facto standard in `std`; OTOH, having to add Cargo features to several crates to be able to filter is not common (it's also error-prone).

(+) yes, `println` is also a new feature that we need to maintain. However, `println` is very close to the existing log filters (an additional log level) so that feels more like an extension that an entire new feature, whereas Cargo feature based filters are quite different from `DEFMT_LOG` IMO

Lotterleben, 2021-08-24:
@jonas-schievink since this is blocking other issues, would it be feasible for you to come up with a Go/Alternative by August 30th?

jonas-schievink, 2021-08-24:
I'm not *against* doing this this, or adding `defmt::println!` in general, but if we do go down this route and remove the Cargo feature based configuration, it would no longer be possible to choose a default logging level (which you can currently do by configuring the right default Cargo features). I have found this fairly useful when working on https://github.com/stm32-rs/bxcan, so we should make sure that we're fine with dropping that.

I don't think that keeping the Cargo features adds much of a maintenance burden, they have currently just "been there" without getting touched by us. They also have a pretty straightforward interaction with `DEFMT_LOG`: if either of them says that a log level is enabled, then it is enabled (ie. they're just `OR`-ed together).

Or is there anything I'm missing here? Does anyone else have opinions on this?

japaric, 2021-08-30:
> They also have a pretty straightforward interaction with DEFMT_LOG: if either of them says that a log level is enabled, then it is enabled (ie. they're just OR-ed together).

I would like to see this interaction specified: what happens when both features are in use (is this even allowed? something else). Perhaps it's straightforward; I haven't had time to think about so I simply don't know what their interaction is.

> Does anyone else have opinions on this?

I'd be concerned about having "more than one way to do the same thing" which *can* be addressed with user facing documentation but that needs to be written.
"When should I prefer X? when should I prefer Y? when should I use both?" are among the things to that need to be documented.

japaric, 2021-08-30:
> Does anyone else have opinions on this?

Another thing to keep in mind re: keeping Cargo-feature-based filter. It's at odds with #255.

If both filters are available and some people start using the env-based one instead of the Cargo-features one then the Cargo-features one becomes less effective because it requires dependencies, which may be third-party, to specify Cargo features which the env-based doesn't need. Meaning that either:
- we tell people to still add Cargo features even if they'll uniquely use the env-filter. This is in direct opposition to #255
- people stop adding Cargo features; this makes the Cargo-features less useful as it won't be able to control the log level of these dependencies (crates)

jonas-schievink, 2021-08-30:
Yeah. Maybe we should just go with #519 / this RFC for 0.3 and figure out how to improve on that later?

japaric, 2021-08-31:
@jonas-schievink could you open a separate issue to discuss the needs of bxcan? so far I understood that you'd like to have a way to specify a log level even if DEFMT_LOG is unset. That *could* be something like `defmt_default_log_level!(Info)` (item-level macro) where that log level is used *unless* overridden with e.g. `DEFMT_LOG=bxcan=none`. But I'd like to know more: do you want this 'default' only for same-repo testing? or also when bxcan is used as a dependency? etc. A separate discussion thread would help identify the desired functionality

japaric, 2021-10-04:
implemented in #569

closed 2021-10-04 by japaric