knurling-rs/defmt #552: Question: Multiple Channels

luctius, 2021-07-30:
The book references being able to use multiple channels. From reading the rtt code, I get the impression this is not implemented.

Is using multiple channels (either manual selection or 1 per priority level) currently implements/working, and if not, is that still a feature that is on the roadmap?

japaric, 2021-08-02:
Hello

> The book references being able to use multiple channels.

The book indicates that's *possible* to implement a lock-free logger that e.g. uses one RTT channel per priority level.

> From reading the rtt code, I get the impression this is not implemented (but I could be very wrong here).

Correct, `defmt-rtt` uses a single channel and a (disable all interrupts) critical section for synchronization so it's not lock-free

> Is using multiple channels (either manual selection or 1 per priority level) currently implements/working, and if not, is that still a feature that is on the roadmap?

There's no feature request issue for it at the moment; so far we have not heard any demand for it. (I think you are the first person to ask about it in the issue tracker)

Urhengulas, 2021-08-19:
@luctius Is your question resolved?

closed 2021-08-19 by luctius