knurling-rs/defmt #554: Work around to calling `write!` in a loop?

jeffy1009, 2021-08-03:
Is there a nice work-around for handling the following situation?
```rust
struct Areas {
   areas: BTreeMap<Area>,
}

impl defmt::Format for Areas {
  fn format(&self, f: defmt::Formatter<'_>) {
    for area in self.areas.iter() {
      defmt::write!(f, {}, area);
    }
  }
```

Other than something like this:
```rust
impl defmt::Format for Areas {
  fn format(&self, f: defmt::Formatter<'_>) {
    match self.areas.len() {
       0 = >,
       1 => defmt::write!(f, {}, self.area[0]),
       2 => defmt::write!(f, {}{}, self.area[0], self.area[1]),
       ...
    }
  }
```

japaric, 2021-08-05:
> Is there a nice work-around for handling the following situation?

Not on 0.2.x. This won't compile with 0.2.x but 0.3.0 (`main` branch) does allow multiple `write!`s in a `Format` implementation so that code does compile with defmt 0.3.0.

(There's another option here to add support for directly formatting iterators with `write!` and/or [debug adapters](https://doc.rust-lang.org/std/fmt/struct.DebugMap.html) for formatting lists, sets and map but those would be new features which are very unlikely to be part of 0.2.x as the focus right now is releasing 0.3.0)

The answer is to wait for the 0.3.0 release or use the `main` branch.

Urhengulas, 2021-08-19:
@jeffy1009 Is your question resolved?

jeffy1009, 2021-08-19:
Yes, you can close it ;)

closed 2021-08-20 by Urhengulas