knurling-rs/defmt #558: defmt-print shows malformed data when streaming from JLinkGDBServer

Bhav97, 2021-08-14:
When I try to pipe RTT data from JLinkGDBServer into defmt-print I get "`Error: malformed data`". 

Setup:
Since the data coming from the GDB server has a header I pipe it into dd first to skip the header bytes like.
`JLinkRTTClient.exe | dd bs=1 skip=1419 | defmt-print -e ./target/thumbv6m-none-eabi/debug/pico-rs`

Observations:
This works well for simple static logs. But when I try to log a formatted string, it errors out with malformed data. 
If  I log the RTT data to a file using  `JLinkRTTClient.exe | dd bs=1 skip=1419 of=log.bin` and then input it to defmt-print like `cat log.bin | defmt-print -e ./target/thumbv6m-none-eabi/debug/pico-rs` then the logs are parsed successfully

As far as I can tell,  formatted logs contain the token and format values and while streaming defmt-print tries to parse the log without reading all the values. 

I have tried this with defmt=0.2.3, defmt-rtt=0.2.0

Lotterleben, 2021-08-16:
Hi, could you share details about how exactly you're setting up your debugging session so we can reproduce your setup and hopefully the bug?

Bhav97, 2021-08-16:
My development setup is in WSL2 Ubuntu20.04, the host machine is Windows 11 Version 21H2 Build 22000.132
I have the latest JLink Software and Documentation Pack (v7.53d beta) with GDB from the ARM GCC pack (10-2020-q4-major)

I start the GDB Server first with
`JLinkGDBServerCL.exe -device RP2040_M0_0 -if swd -speed 4000K -noir -nolocalhostonly`

And then I use the following launch configuration in vscode
```json
{
	"version": "0.2.0",
	"configurations": [
		{
			"name": "Cortex Debug",
			"cwd": "${workspaceRoot}",
			"executable": "./target/thumbv6m-none-eabi/debug/pico-rs",
			"request": "launch",
			"type": "cortex-debug",
			"servertype": "external",
			"gdbTarget": "192.168.0.159:2331",
			"runToMain": true,
			"svdFile": "./extra/rp2040.svd",
			"postLaunchCommands": [
				"set mem inaccessible-by-default off",
				"mon reset 0",
			]
		}
	]
}
```

I believe a simple project with the following snippet should be able to reproduce the issue
```rust
fn main() -> ! {
    loop {
        cortex_m::asm::delay(1_000_000);
        defmt::info!("Loop!");
        let pr: u32 = 0xdeadbeef;
        defmt::info!("Hello! {}", pr);
    }
}
```

Lotterleben, 2021-08-17:
Hi,
it seems that this is an issue with the J-Link tools, so unfortunately we can't fix this :(.
*However*, with the upcoming 0.3.0 release, we'll introduce framing, which will be able to recover from malformed data (and will thus also automatically throw away the gdb header you're currently skipping manually).

closed 2021-08-17 by Lotterleben