knurling-rs/defmt #565: decode breaks when implemening Format

devsnek, 2021-08-29:
when i implement this trait, i get an error from defmt:

```rs
impl defmt::Format for State {
    fn format(&self, fmt: defmt::Formatter) {
        use chrono::TimeZone;
        let dt = chrono::Utc
            .ymd_opt(self.year, self.month, self.day)
            .and_hms_opt(self.h, self.m, self.s)
            .map(|dt| dt + chrono::Duration::seconds(self.tz_offset as i64))
            .single();

        defmt::write!(
            fmt,
            "FIX: {} | SVs: {} | HDOP: {} | {}",
            self.fix,
            self.svs,
            self.hdop,
            match dt {
                Some(s) => s.to_string(),
                None => "INVALID".to_owned(),
            },
        );
    }
}
```

```
 INFO  FIX: 0 
└─ pico_clock::__cortex_m_rt_main @ src/main.rs:293
(HOST) ERROR failed to decode defmt data: [0, 0, 0, 0, 2, 0, 0, 0, 0, 3, 7, 49, 4e, 56, 41, 4c, 49, 44]
Error: malformed data
```

However, if I use `info!` directly, it works fine:

```rs
impl State {
    fn print(&self) {
        use chrono::TimeZone;
        let dt = chrono::Utc
            .ymd_opt(self.year, self.month, self.day)
            .and_hms_opt(self.h, self.m, self.s)
            .map(|dt| dt + chrono::Duration::seconds(self.tz_offset as i64))
            .single();

        info!(
            "FIX: {} | SVs: {} | HDOP: {} | {}",
            self.fix,
            self.svs,
            self.hdop,
            match dt {
                Some(s) => s.to_string(),
                None => "INVALID".to_owned(),
            },
        );
    }
}
```

```
 INFO  FIX: 0 | SVs: 0 | HDOP: 0.0 | INVALID
└─ pico_clock::__cortex_m_rt_main @ src/main.rs:293
 INFO  FIX: 0 | SVs: 0 | HDOP: 0.0 | INVALID
└─ pico_clock::__cortex_m_rt_main @ src/main.rs:293

// ... and so on
```

Urhengulas, 2021-09-01:
Hello @devsnek,

Thank you for your report. What version of `probe-run` are you using?
And which exact version of `defmt` and `defmt-rtt`? You can check this in your `Cargo.lock`.

devsnek, 2021-09-01:
defmt is 0.2.3, defmt is 0.2.0. probe-run is from https://github.com/rp-rs/probe-run

Urhengulas, 2021-09-01:
Just for my understanding: Is your code `no_std`? Cause you are using `.to_string`.

Urhengulas, 2021-09-01:
Okay, I am able to reproduce it in `no_std` anyways.

devsnek, 2021-09-01:
my code is `no_std` but i have alloc enabled, which includes `alloc::string::ToString`

Urhengulas, 2021-09-02:
It's quite weird. It seems that `write!` generally doesn't like `|`:

## findings

### `write!(fmt, "|")` breaks, but doesn't report error
```rust
struct State;
impl Format for State {
    fn format(&self, fmt: defmt::Formatter) {
        write!(fmt, "|")
    }
}
```

#### Output
Although it says there is no error, it doesn't print the `|` and the call stack is corrupted.

```
(HOST) INFO  flashing program (7.21 KiB)
(HOST) INFO  success!
────────────────────────────────────────────────────────────────────────────────
────────────────────────────────────────────────────────────────────────────────
stack backtrace:
   0: lib::inline::__bkpt
        at ./asm/inline.rs:13:5
   1: __bkpt
        at ./asm/lib.rs:49:17
   2: defmt_app::exit
        at src/lib.rs:26:5
   3: hello::__cortex_m_rt_main
        at src/bin/hello.rs:17:5
   4: main
        at src/bin/hello.rs:14:1
   5: Reset
(HOST) WARN  call stack was corrupted; unwinding could not be completed
(HOST) INFO  device halted without error
```

### `write!(fmt, "|{}", 10)` breaks
```rust
struct State;
impl Format for State {
    fn format(&self, fmt: defmt::Formatter) {
        write!(fmt, "|{}", 10)
        //                 ^^ can be any other number
    }
}
```

#### Output
```
(HOST) INFO  flashing program (7.30 KiB)
(HOST) INFO  success!
────────────────────────────────────────────────────────────────────────────────
(HOST) ERROR failed to decode defmt data: [3, 0, 2, 1, a, 0, 0, 0]
Error: malformed data
```

### `write!(fmt, "|{}", 10)` breaks, but different
```rust
struct State;
impl Format for State {
    fn format(&self, fmt: defmt::Formatter) {
        write!(fmt, "{}|", 10)
        //                 ^^ can be any other number
    }
}
```

#### Output
```
(HOST) INFO  flashing program (7.30 KiB)
(HOST) INFO  success!
────────────────────────────────────────────────────────────────────────────────
0 INFO  
└─ hello::__cortex_m_rt_main @ src/bin/hello.rs:19
(HOST) ERROR failed to decode defmt data: [a, 0, 0, 0]
Error: malformed data
```

Urhengulas, 2021-09-02:
@japaric:
>`|` is used in defmt internals for encoding enums; it shouldn't be used in user-facing `write!`
>
> user-facing `write!` should reject format strings that contain `|` or kind of escape them so that the decoder doesn't interpret them as enums

Also we found that this issue was raised before: https://github.com/knurling-rs/defmt/issues/210

Dirbaio, 2021-09-02:
Rejecting `|` in `write!` calls is a lousy fix.

Since symbol names are JSON, why not encode whether an interned index is an enum or not using JSON? i.e instead of `None|Some({})`, store something like `{"type": "enum", "variants": ["None", "Some({})"]}`.

Urhengulas, 2021-09-05:
> Rejecting `|` in `write!` calls is a lousy fix.
> 
> Since symbol names are JSON, why not encode whether an interned index is an enum or not using JSON? i.e instead of `None|Some({})`, store something like `{"type": "enum", "variants": ["None", "Some({})"]}`.

Interesting idea! Will look into that :)

Lotterleben, 2021-09-06:
alternative idea from @jonas-schievink : mark the defmt-internal Strings that use `|` as such [this marker already exists], only split on `|` if this marker is set. this way we wouldn't break the wire format (we wouldn't need to change the JSON string encoding).

japaric, 2021-10-15:
fixed in #614

closed 2021-10-15 by japaric