knurling-rs/defmt #575: using `probe-run` with `encoding-rzcobs` results in decoding error

japaric, 2021-09-13:
## Steps to reproduce

1. Instantiate `app-template` and configure it to use the git version of `defmt`. I used `main` branch (59d2074)
2. Clone `probe-run`, edit `Cargo.toml` to use the `main` branch of `defmt-decoder`. NOTE remove the `version` field from the `defmt-decoder` dep to eliminate ambiguity

``` console
$ cargo run --bin hello
(HOST) INFO  flashing program (6.57 KiB)
(HOST) INFO  success!
────────────────────────────────────────────────────────────────────────────────
────────────────────────────────────────────────────────────────────────────────
(HOST) INFO  device halted without error
```

``` console
$ cargo run --bin levels
(HOST) INFO  flashing program (6.70 KiB)
(HOST) INFO  success!
────────────────────────────────────────────────────────────────────────────────
(HOST) ERROR failed to decode defmt data: [0, 2, 7e, 0, 3, 1, 7a, 0, 4, 2, 7a, 0, 5, 3, 7a, 0, 6, 4, 7a, 0]
Error: malformed data
```

Enabling defmt's `encoding-raw` feature makes both examples work

justahero, 2021-09-20:
Maybe related to #559?

japaric, 2021-09-27:
I think I see the problem: `probe-run` is using the old [`decode`](https://github.com/knurling-rs/probe-run/blob/fd606d6f90b863190c2a0170bd7262150595727c/src/main.rs#L299) API instead of the new `stream_decoder` API like [qemu-run](https://github.com/knurling-rs/defmt/blob/e021e7d013b9202f70bc762f6909401f85142c89/qemu-run/src/main.rs#L67) does

closed 2021-09-27 by bors[bot]