knurling-rs/defmt #625: Dual versions of `defmt` causes `links` error when it should not

korken89, 2021-11-10:
Hi all,

I'm looking to add to all the crates I maintain so it support `defmt 0.2` and `0.3`, but I get a `links` error even though the dependencies are optional.
If you do the same for `cortex-m` which also uses the `links` argument it does however work!

Something is being done in `defmt` which breaks optional dependencies - not sure what.
This does however block us from supporting both `0.2` and `0.3`, which would allow us to do a non-breaking release of crates with `0.3` support.

**Not** working `Cargo.toml` (run in empty project):

```toml
[package]
name = "dual-defmt"
version = "0.1.0"
edition = "2021"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]

[dependencies.defmt02]
package = "defmt"
version = "0.2"
optional = true

[dependencies.defmt03]
package = "defmt"
version = "0.3"
optional = true
```

Working `Cargo.toml` (run in empty project):

```toml
[package]
name = "dual-defmt"
version = "0.1.0"
edition = "2021"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]

[dependencies.defmt02]
package = "cortex-m"
version = "0.6"
optional = true

[dependencies.defmt03]
package = "cortex-m"
version = "0.7"
optional = true
```

jonas-schievink, 2021-11-10:
Duplicate of https://github.com/knurling-rs/defmt/issues/426, I believe

korken89, 2021-11-11:
Seems like it, that's unfortunate.
I don't want to release new major versions just due to this, but maybe there's no way around it.

korken89, 2021-11-11:
Here is the root `cargo` issue as well: https://github.com/rust-lang/cargo/issues/5969

korken89, 2021-11-11:
Are there any issues with specifying a range of versions for `defmt` for libraries?
E.g.:

```toml
[dependencies.defmt]
version = ">=0.2,<0.4"
optional = true
```

This seems to work in my experiments in `heapless`, but I'm not sure if this has unintended consequences.

Urhengulas, 2021-12-01:
Hi @korken89, let's move the discussion to #426, in order to scatter it not too much.

closed 2021-12-01 by Urhengulas