knurling-rs/defmt #629: versions in "version mismatch" error message are backwards

japaric, 2021-11-22:
output is

``` text
Error: defmt wire format version mismatch: firmware is using 0.2, `probe-run` supports 3
suggestion: `cargo install` a different version of `probe-run` that supports defmt 3
```

last line should say "supports defmt 0.2"

jonathanpallant, 2021-11-22:
This error message comes from `decoder::elf2table::check_version`.

jonathanpallant, 2021-11-22:
This issue appears to be fixed in https://github.com/knurling-rs/defmt/commit/89843557a40163ca6e1a3fbed5e890223e14628a.

jonathanpallant, 2021-11-22:
OK, so the fix is in `main` but was made after the defmt-0.3.0 release. To get this out, do we do a defmt-0.3.1 release (bumping all crates versions, even if there are no changes), or do a defmt-decoder-0.3.1 release and leave everything else as-is?

Because we have semver (and use it correctly?!) it feels like we should do that latter; there seems to be no sense in bumping a bunch of crates with no changes in them just to keep the version numbers in sync.

Urhengulas, 2021-11-23:
Let's close this issue (since it is solved on main) and move the release discussion to zulip 😁