knurling-rs/defmt #632: Negative values for i16 formatted as hexadecimals have enough FFs for an i128.

mehmetalianil, 2021-11-24:
When I print out a i16 as an i16 as in here:
```
defmt::debug!("x: {=i16:#x},y: {=i16:#x},z: {=i16:#x}", 
                x,
                y,
                z);
```
I get a twos complement value, but for negatives, I get enough `f`s for an i128:
```
DEBUG x: 0xfffffffffffffffffffffffffffffeda,y: 0xfffffffffffffffffffffffffffffd30,z: 0x40b3
```

I don't say that it is a bug, but could be deceiving.

Urhengulas, 2021-12-01:
@mehmetalianil Thank you for your issue report.

I have a slight problem understanding the problem. Could you help me understand it please? Therefore it might be helpful if you could elaborate your code example with the input values you used (the types and values for x, y, z).

> When I print out a i16 as an i16

Did you mean to use `i16` two times, or is one of them meant to be a different number type?

justahero, 2021-12-01:
Hello @Urhengulas,

the following steps I used to reproduce the behaviour.

Create a new project via [app-template](https://github.com/knurling-rs/app-template) repository.
Then replace the content of one of the programs, e.g. `hello.rs`, with the following code

```rust
#![no_main]
#![no_std]

use defmt_test_example as _; // global logger + panicking-behavior + memory layout

#[cortex_m_rt::entry]
fn main() -> ! {
    defmt::info!("x: {=i16:#x}", -1_i16);
    defmt_test_example::exit()
}
```

Run the program with `DEFMT_LOG=info cargo rb hello`, the generated output is as follows:

```
0 INFO  x: 0xffffffffffffffffffffffffffffffff
└─ hello::__cortex_m_rt_main @ src/bin/hello.rs:8
```

The issue here is, the `i16` value is represented with the [`Arg::Ixx`](https://github.com/knurling-rs/defmt/blob/main/decoder/src/lib.rs#L282) variant. Unfortunately when the resulting `Frame` for hex values is formatted in [Frame::format_i128](https://github.com/knurling-rs/defmt/blob/1bd226b03e80d02bb02064bfcdac36291aa2f52a/decoder/src/frame.rs#L261-L289) the integer is treated as a `i128` type. The displayed value is not handled as `-1_i16` but rather as `-1_i128`, therefore the hex value is filled with `f`.

Urhengulas, 2021-12-01:
I see. Thank you @justahero! 😄

mehmetalianil, 2021-12-01:
Yep, much clearer and reproducible explanation of the issue there. (: 
Thanks, @justahero.

Urhengulas, 2021-12-14:
Fixed by #646

closed 2021-12-14 by Urhengulas

justahero, 2021-12-14:
This still requires a new release of `probe-run` as this uses the `defmt-decoder` crate. The fix should be available in the next release of `probe-run`.