knurling-rs/defmt #637: defmt_rtt can panic on multicore

jannic, 2021-11-25:
The code at https://github.com/knurling-rs/defmt/blob/964c8e7457bd9cd3d7bd019e20f6e3697968debe/firmware/defmt-rtt/src/lib.rs#L50-L52 is not multi-core safe: If two cores both try to acquire the lock, the second one will panic.

Dirbaio, 2021-11-26:
defmt-rtt assumes singlecore, it uses `interrupt::disable()` like `cortex_m::interrupt::free()`. 

You'd have to change it to use a custom critical section for your chip (RP2040 I guess?). I have a version of `defmt-rtt` that uses `critical-section` [here](https://github.com/embassy-rs/nrf-softdevice/blob/master/nrf-softdevice-defmt-rtt/src/lib.rs#L39), together with the impl here https://github.com/rp-rs/rp-hal/pull/151 it should work.

Perhaps changing `defmt-rtt` to use critical-section would be a welcome addition to upstream defmt-rtt?

jannic, 2021-11-26:
Ok, that makes sense. However I wonder why TAKEN is an atomic, if it's only used on single core while interrupts are disabled.

Yes, my use case is rp2040, how did you guess? :-)

jonathanpallant, 2021-11-26:
Can defmt be generic over some type `T: DefmtLock`? You can have a default implementation for systems with atomic values, and a user can substitute in an implementation that uses a stronger lock (e.g. RP2040 has spin-locks) without having to fork the code.

jannic, 2021-11-26:
If defmt-rtt used `critical-section` like in the version @Dirbaio mentioned, individual firmwares could replace the actual implementation using the `custom-impl` feature of `critical-section`. It would not be necessary for defmt-rtt to directly support different implementations.
Not sure if the additional dependency is justified, though.

Dirbaio, 2021-11-26:
> @jannic  Ok, that makes sense. However I wonder why TAKEN is an atomic, if it's only used on single core while interrupts are disabled.

I think it could be a `static mut` as well, but relaxed atomics require less unsafe and have no extra overhead.

> @jonathanpallant  Can defmt be generic over some type T: DefmtLock? 

Unfortunately the Logger is global, it can't have generics. `defmt-rtt` could offer some macro for you to set your own lock impl, but then the UX is the same as `critical-section`.

jannic, 2022-02-10:
Fixed by https://github.com/knurling-rs/defmt/pull/640

closed 2022-02-10 by jannic