knurling-rs/defmt #653: `Strip` in `release` with defmt

mryndzionek, 2022-01-19:
In `release` I would like to use `strip` and remove `defmt` entirely, but then I get:
```rs
Error: `.defmt` section found, but no version symbol - check your linker configuration
```

jonas-schievink, 2022-01-20:
`strip` has a `--remove-section` flag, have you tried that?

mryndzionek, 2022-01-20:
It turns out that `strip` works okay, but the Error is from `probe-run`...

mryndzionek, 2022-01-20:
I made some progress using `--only-keep-debug`, but now `probe-run` returns:
```
Error: vector table section is too short. (has length: 0 - should be at least 16)
```

jonas-schievink, 2022-01-20:
`--only-keep-debug` is the opposite of stripping debug info, it removes all non-debug data

mryndzionek, 2022-01-23:
Closing this for now. It seems that stripping won't get me much in `--release`

closed 2022-01-24 by Urhengulas