knurling-rs/defmt #654: Host side timestamps

mryndzionek, 2022-01-23:
Is this supported already? Is this repo the right place to implement this?

Urhengulas, 2022-01-24:
Hi @mryndzionek, Thank you for your question!

The output of `probe-run` (our host tool to display defmt logs) currently doesn't display host timestamps and also the `defmt-decoder` library doesn't provide the host timestamp information when building your own software on top.

If you are building your own software on top of `defmt-decoder` it is rather easy for you to obtain your own host timestamps (e.g. using [`chrono::Utc::now()`](https://docs.rs/chrono/latest/chrono/offset/struct.Utc.html#method.now)) and using it together with the information from defmt.

On a side-note: I am currently working on https://github.com/knurling-rs/defmt/pull/651, which is structured output for `probe-run`, which also includes the host_timestamp. Please send feedback if you feel like.

Greetings 👋🏾

japaric, 2022-03-24:
`probe-run` v0.3.3 has a `--json` flag to emit the logs in JSON format. each log frame has a `host_timestamp` field which is the timestamp at which the host received the frame (!= the time at which the target emitted it; there's a variable delay due to USB communication). the timestamp is in seconds from the unix timestamp (you can use `chrono::Utc.timestamp(host_timestamp, 0)` to get a `DateTime`). The `--json` feature is documented in the book: https://defmt.ferrous-systems.com/json-output.html

mryndzionek, 2022-04-04:
> With `probe-run v0.3.3` supporting this I suppose we can close this, or are there questions left @mryndzionek?

Yes, we can close this. Thank you!

closed 2022-04-04 by mryndzionek