knurling-rs/defmt #664: Change output format on host

timhoeppner, 2022-03-09:
I'm not really sure if this is considered an issue but I can't seem to find a way to configure the format shown on the host. Its currently shown like:

```
0 Hello, world!
└─ minimal::app::init @ src\bin\minimal.rs:32
1 in foo
└─ minimal::app::foo @ src\bin\minimal.rs:77
2 in bar
└─ minimal::app::bar @ src\bin\minimal.rs:86
3 in foo
└─ minimal::app::foo @ src\bin\minimal.rs:77
```

I might like to do something like:

```
minimal::app::init[src\bin\minimal.rs:32]: 0 Hello, world!
minimal::app::foo[src\bin\minimal.rs:77]: 1 in foo
minimal::app::bar[src\bin\minimal.rs:86]: 2 in bar
minimal::app::foo[src\bin\minimal.rs:77]: 3 in foo
```

or any other permutation.

Is this possible?

japaric, 2022-03-21:
`defmt` does not define the output format but if you are using `probe-run` you can use the new-ish `--json` flag (it's there in v0.3.3) and then pipe that into some formatter program that produces the desired format, i.e. `probe-run --chip some-chip --json my-elf | my-formatter`. you can use `defmt-json-schema` to decode `probe-run`'s output. there's example output and an example formatter in https://github.com/knurling-rs/defmt/tree/a10d971661d723e8e58e24f4be51ce5f1767d6df/decoder/defmt-json-schema/examples

timhoeppner, 2022-03-21:
This is great, thanks!

closed 2022-03-21 by timhoeppner

timhoeppner, 2022-03-31:
I'm putting together a little tool to process the json stream from probe-run if anyone is interested. Eventually plan to have a custom definable format but just slapped this together to satisfy my immediate need.
https://github.com/timhoeppner/probe-run-defmt-fmt