knurling-rs/defmt #673: error LNK2019: unresolved external symbol _defmt_panic referenced in function

haata, 2022-05-23:
This is likely similar to #411 

I have a reproducible case (only affects Windows).

- Nightly (cargo 1.63.0-nightly (a4c1cd0eb 2022-05-18))
- https://github.com/kiibohd/kiibohd-core/tree/main/kll-core (exact hash https://github.com/kiibohd/kiibohd-core/tree/528672a0f7f255eb95cda7fd5423cfc553fa959e)
- cargo run works
- cargo test
```
  = note: libkll_hid-48b2e4c1faf634c7.rlib(kll_hid-48b2e4c1faf634c7.8ek79hmbmne36vd.rcgu.o) : error LNK2019: unresolved external symbol _defmt_panic referenced in function _ZN57_$LT$kll_hid..Locale$u20$as$u20$defmt..traits..Format$GT$6format17h16cbf08301323530E
          libkll_hid-48b2e4c1faf634c7.rlib(kll_hid-48b2e4c1faf634c7.8ek79hmbmne36vd.rcgu.o) : error LNK2019: unresolved external symbol _defmt_acquire referenced in function _ZN57_$LT$kll_hid..Locale$u20$as$u20$defmt..traits..Format$GT$6format17h16cbf08301323530E
          libdefmt-84a1edf1a654ad5f.rlib(defmt-84a1edf1a654ad5f.defmt.6fd46b78-cgu.8.rcgu.o) : error LNK2001: unresolved external symbol _defmt_acquire
          libkll_hid-48b2e4c1faf634c7.rlib(kll_hid-48b2e4c1faf634c7.8ek79hmbmne36vd.rcgu.o) : error LNK2019: unresolved external symbol _defmt_release referenced in function _ZN57_$LT$kll_hid..Locale$u20$as$u20$defmt..traits..Format$GT$6format17h16cbf08301323530E
          libdefmt-84a1edf1a654ad5f.rlib(defmt-84a1edf1a654ad5f.defmt.6fd46b78-cgu.8.rcgu.o) : error LNK2001: unresolved external symbol _defmt_release
          libdefmt-84a1edf1a654ad5f.rlib(defmt-84a1edf1a654ad5f.defmt.6fd46b78-cgu.8.rcgu.o) : error LNK2019: unresolved external symbol _defmt_flush referenced in function _ZN5defmt5flush17h7ba29d74f1244836E
          libdefmt-84a1edf1a654ad5f.rlib(defmt-84a1edf1a654ad5f.defmt.6fd46b78-cgu.3.rcgu.o) : error LNK2019: unresolved external symbol _defmt_write referenced in function _ZN5defmt6export3f3217hed4f57cccfafeed4E
          libdefmt-84a1edf1a654ad5f.rlib(defmt-84a1edf1a654ad5f.defmt.6fd46b78-cgu.6.rcgu.o) : error LNK2001: unresolved external symbol _defmt_write
          libdefmt-84a1edf1a654ad5f.rlib(defmt-84a1edf1a654ad5f.defmt.6fd46b78-cgu.3.rcgu.o) : error LNK2019: unresolved external symbol _defmt_timestamp referenced in function _ZN5defmt6export6header17h47dc426ec968adf6E
          C:\Users\tripl\source\repos\kiibohd\kiibohd-core\target\debug\deps\kll_core-bd87741b11f4a13f.exe : fatal error LNK1120: 6 unresolved externals
```

I likely can resolve this if I make defmt an optional opt-in dependency.

haata, 2022-05-23:
An even smaller example (same git repo, also only a problem on Windows).

- https://github.com/kiibohd/kiibohd-core/tree/528672a0f7f255eb95cda7fd5423cfc553fa959e/kll-hid
- cargo test
```
  = note: libkll_hid-48b2e4c1faf634c7.rlib(kll_hid-48b2e4c1faf634c7.8ek79hmbmne36vd.rcgu.o) : error LNK2019: unresolved external symbol _defmt_panic referenced in function _ZN57_$LT$kll_hid..Locale$u20$as$u20$defmt..traits..Format$GT$6format17h16cbf08301323530E
          libkll_hid-48b2e4c1faf634c7.rlib(kll_hid-48b2e4c1faf634c7.8ek79hmbmne36vd.rcgu.o) : error LNK2019: unresolved external symbol _defmt_acquire referenced in function _ZN57_$LT$kll_hid..Locale$u20$as$u20$defmt..traits..Format$GT$6format17h16cbf08301323530E
          libdefmt-84a1edf1a654ad5f.rlib(defmt-84a1edf1a654ad5f.defmt.6fd46b78-cgu.8.rcgu.o) : error LNK2001: unresolved external symbol _defmt_acquire
          libkll_hid-48b2e4c1faf634c7.rlib(kll_hid-48b2e4c1faf634c7.8ek79hmbmne36vd.rcgu.o) : error LNK2019: unresolved external symbol _defmt_release referenced in function _ZN57_$LT$kll_hid..Locale$u20$as$u20$defmt..traits..Format$GT$6format17h16cbf08301323530E
          libdefmt-84a1edf1a654ad5f.rlib(defmt-84a1edf1a654ad5f.defmt.6fd46b78-cgu.8.rcgu.o) : error LNK2001: unresolved external symbol _defmt_release
          libdefmt-84a1edf1a654ad5f.rlib(defmt-84a1edf1a654ad5f.defmt.6fd46b78-cgu.8.rcgu.o) : error LNK2019: unresolved external symbol _defmt_flush referenced in function _ZN5defmt5flush17h7ba29d74f1244836E
          libdefmt-84a1edf1a654ad5f.rlib(defmt-84a1edf1a654ad5f.defmt.6fd46b78-cgu.3.rcgu.o) : error LNK2019: unresolved external symbol _defmt_write referenced in function _ZN5defmt6export3f3217hed4f57cccfafeed4E
          libdefmt-84a1edf1a654ad5f.rlib(defmt-84a1edf1a654ad5f.defmt.6fd46b78-cgu.6.rcgu.o) : error LNK2001: unresolved external symbol _defmt_write
          libdefmt-84a1edf1a654ad5f.rlib(defmt-84a1edf1a654ad5f.defmt.6fd46b78-cgu.3.rcgu.o) : error LNK2019: unresolved external symbol _defmt_timestamp referenced in function _ZN5defmt6export6header17h47dc426ec968adf6E
          C:\Users\tripl\source\repos\kiibohd\kiibohd-core\target\debug\deps\kll_core-bd87741b11f4a13f.exe : fatal error LNK1120: 6 unresolved externals
          ```

Urhengulas, 2022-05-31:
Thank you for reporting, we will look into it.

KodrAus, 2022-10-09:
> defmt currently does not support hosted environments like Linux, Windows, macOS, etc. this can be a bit of an issue when using / (host) testing libraries that use defmt

Is there any advice on how/whether we should implement `defmt::Format` in libraries? We've got a PR over in `uuid` https://github.com/uuid-rs/uuid/pull/639 that adds it, but have run into what I think is this issue in our CI with `defmt::write!`. We can work around it of course, but I've been wanting to keep our `--all-features` builds working on tier 1 platforms.

jhillyerd, 2022-10-13:
I am running into link errors for a shared library that uses `derive(Format)`, and no logging statements.  The library contains structs that are used by my MCU, Linux, and Windows.  It builds fine on Linux, but on Windows I get:

```
  = note: libshared-7e68f9b1edf48cc2.rlib(shared-7e68f9b1edf48cc2.xb1jn04cva8eh78.rcgu.o) : error LNK2019: unresolved external symbol _defmt_acquire referenced in function _ZN67_$LT$shared..message..FromHost$u20$as$u20$defmt..traits..Format$GT$6format17h0b99be05839c96deE
          libdefmt-021a50e4f73e2db0.rlib(defmt-021a50e4f73e2db0.defmt.ee3c75f0-cgu.9.rcgu.o) : error LNK2001: unresolved external symbol _defmt_acquire
          libshared-7e68f9b1edf48cc2.rlib(shared-7e68f9b1edf48cc2.xb1jn04cva8eh78.rcgu.o) : error LNK2019: unresolved external symbol _defmt_release referenced in function _ZN67_$LT$shared..message..FromHost$u20$as$u20$defmt..traits..Format$GT$6format17h0b99be05839c96deE
          libdefmt-021a50e4f73e2db0.rlib(defmt-021a50e4f73e2db0.defmt.ee3c75f0-cgu.9.rcgu.o) : error LNK2001: unresolved external symbol _defmt_release
          libshared-7e68f9b1edf48cc2.rlib(shared-7e68f9b1edf48cc2.xb1jn04cva8eh78.rcgu.o) : error LNK2019: unresolved external symbol _defmt_panic referenced in function _ZN67_$LT$shared..message..FromHost$u20$as$u20$defmt..traits..Format$GT$6format17h0b99be05839c96deE
          libdefmt-021a50e4f73e2db0.rlib(defmt-021a50e4f73e2db0.defmt.ee3c75f0-cgu.9.rcgu.o) : error LNK2019: unresolved external symbol _defmt_flush referenced in function _ZN5defmt5flush17h6023f786f51632fdE
          libdefmt-021a50e4f73e2db0.rlib(defmt-021a50e4f73e2db0.defmt.ee3c75f0-cgu.3.rcgu.o) : error LNK2019: unresolved external symbol _defmt_write referenced in function _ZN5defmt6export3f3217h160241a35c1e8599E
          libdefmt-021a50e4f73e2db0.rlib(defmt-021a50e4f73e2db0.defmt.ee3c75f0-cgu.6.rcgu.o) : error LNK2001: unresolved external symbol _defmt_write
          libdefmt-021a50e4f73e2db0.rlib(defmt-021a50e4f73e2db0.defmt.ee3c75f0-cgu.3.rcgu.o) : error LNK2019: unresolved external symbol _defmt_timestamp referenced in function _ZN5defmt6export6header17ha728642f8139efcdE
          C:\Users\james\Documents\GitHub\hw-gauge\daemon\target\debug\deps\hw_gauge_winsvc.exe : fatal error LNK1120: 6 unresolved externals
```

I'm not finding enough information online to know if this is an error on my part, or similar to reports above.

thejpster, 2022-12-11:
I just fell over this issue. I have a library where I derive `defmt::Format` for a type, but also want to run unit tests.

```text
  = note: libdefmt-070f26dc2660ddc9.rlib(defmt-070f26dc2660ddc9.defmt.167e37fa-cgu.12.rcgu.o) : error LNK2019: unresolved external symbol _defmt_acquire referenced in function _ZN5defmt5flush17h2fc9ebaccd76d845E
          libdefmt-070f26dc2660ddc9.rlib(defmt-070f26dc2660ddc9.defmt.167e37fa-cgu.12.rcgu.o) : error LNK2019: unresolved external symbol _defmt_flush referenced in function _ZN5defmt5flush17h2fc9ebaccd76d845E
          libdefmt-070f26dc2660ddc9.rlib(defmt-070f26dc2660ddc9.defmt.167e37fa-cgu.12.rcgu.o) : error LNK2019: unresolved external symbol _defmt_release referenced in function _ZN5defmt5flush17h2fc9ebaccd76d845E
          libdefmt-070f26dc2660ddc9.rlib(defmt-070f26dc2660ddc9.defmt.167e37fa-cgu.1.rcgu.o) : error LNK2019: unresolved external symbol _defmt_write referenced in function _ZN5defmt6export3f3217hc6d5ce75e66caeb7E
          libdefmt-070f26dc2660ddc9.rlib(defmt-070f26dc2660ddc9.defmt.167e37fa-cgu.6.rcgu.o) : error LNK2001: unresolved external symbol _defmt_write
          libdefmt-070f26dc2660ddc9.rlib(defmt-070f26dc2660ddc9.defmt.167e37fa-cgu.1.rcgu.o) : error LNK2019: unresolved external symbol _defmt_timestamp referenced in function _ZN5defmt6export6header17he32352bb0799f419E
          C:\Users\msn\Documents\GitHub\Neotron-BMC\target\debug\deps\neotron_bmc_protocol-4f03f65c89f102c7.exe : fatal error LNK1120: 5 unresolved externals
```

kpishere, 2023-05-22:
Ah!  I resolved this error by actually using the feature.  Forcing the linking/loading of the timestamp function, I suppose.

```
defmt::timestamp!("Starting MyProgram...");
```

Urhengulas, 2023-05-22:
The default should be that timestamps are turned off: https://github.com/knurling-rs/defmt/blob/aceeaa07e09822cbf07efcd6b8fb41d09e8f0ee2/defmt/src/lib.rs#L361-L364

Does this error also happen to you on stable?

kpishere, 2023-05-22:
@Urhengulas My project has no chance of building on stable.  I'm doing no_std no-OS on RP2040 microcontroller, project will not build without nightly.

barafael, 2023-10-27:
I have similar errors. I am building a server that serializes and deserializes data according to a protocol defined in a crate which is shared between linux, the MCU, and windows. This protocol crate derives defmt::Format for a bunch of types.

I get these missing symbols:

* `_defmt_write`
* `_defmt_timestamp`
* `_defmt_acquire`
* `_defmt_flush`
* `_defmt_release`

thejpster, 2023-10-28:
@barafael: https://github.com/knurling-rs/defmt/issues/673#issuecomment-1142150012 is probably the answer you need.

barafael, 2023-10-28:
@thejpster thanks for the hint, this does not work though. I get the same error and these imports are unused - the library only uses `defmt::Format`, not `info!` etc.

barafael, 2023-10-28:
For reference, here's a repo to reproduce:
https://github.com/barafael/defmt-on-windows-issue

thejpster, 2023-10-28:
AFAIK have to use cfg_attr to turn off anything defmt related. The whole job of the library is to hide strings in your debug info using linker magic and that is just incompatible with a using a Real Operating System. 

See https://crates.io/crates/embedded-sdmmc for a library I wrote that uses log or defmt, using feature flags.

barafael, 2023-10-28:
Thanks @thejpster, that did the trick.