knurling-rs/defmt #680: defmt-print: exit when stdin is closed

kellda, 2022-06-17:
When the standard input is closed, `defmt-print` uses 100% CPU by reading 0 bytes from stdin in a loop. I think the most useful behaviour is to exit once all received logs are print. This allows to use e.g. `nc -d localhost 8765 | defmt-print -e <elf> &` and to have `defmt-print` exit when the server is restarted.

https://github.com/knurling-rs/defmt/blob/7dff53f6c7c423b0d2c0287687b6882b3cbc1f7b/print/src/main.rs#L70-L73

japaric, 2022-06-21:
breaking the loop on `Ok(0)` / EOF sounds good to me as this is stdin (might not be ok with `File` and named pipes but that's not the case here)

Urhengulas, 2022-06-24:
I am unsure what the value of `n` (in `Ok(n)`) is, if the tcp server currently does not send data, but not since it is shut down, but just because there isn't currently any event which gets logged. It probably is not `0`, or is it?

Reading [the docs](https://doc.rust-lang.org/stable/std/io/trait.Read.html#tymethod.read) it sounds like `Ok(0)` indicates a closed connection for TcpStream:
> If n is 0, then it can indicate one of two scenarios:
> 1. This reader has reached its “end of file” and will likely no longer be able to produce bytes. Note that this does not mean that the reader will always no longer be able to produce bytes. As an example, on Linux, this method will call the recv syscall for a [TcpStream](https://doc.rust-lang.org/stable/std/net/struct.TcpStream.html), where returning zero indicates the connection was shut down correctly. While for [File](https://doc.rust-lang.org/stable/std/fs/struct.File.html), it is possible to reach the end of file and get zero as result, but if more data is appended to the file, future calls to read will return more data.
> 2. The buffer specified was 0 bytes in length.

kellda, 2022-06-24:
Note that defmt-print just read from stdin (the TCP part is handled by `nc` in my example, and the data could as well come from an UART or whatever). Therefore no `TcpStream` is involved.

In any case, when no data is available yet, it just blocks until there is.

commit 7886fa03 (2022-06-28): Merge #682

682: Quit when EOF is reached r=Urhengulas a=kellda

Closes #680
PR #682 (2022-06-28): Quit when EOF is reached
Closes #680