knurling-rs/defmt #691: Code size when using `DEFMT_LOG`

quentinmit, 2022-08-14:
I was trying to figure out why my binary had gotten so huge when I realized that `DEFMT_LOG` is causing a massive blowup in code size. With `DEFMT_LOG=info,discobiker::actors::display=trace`, `cargo expand` shows that every single defmt macro invocation in my package is compiling this `check` function:

```rust
                        const fn check() -> bool {
                            let module_path = "discobiker".as_bytes();
                            if if 27usize > module_path.len() {
                                false
                            } else {
                                module_path[0usize] == 100u8
                                    && module_path[1usize] == 105u8
                                    && module_path[2usize] == 115u8
                                    && module_path[3usize] == 99u8
                                    && module_path[4usize] == 111u8
                                    && module_path[5usize] == 98u8
                                    && module_path[6usize] == 105u8
                                    && module_path[7usize] == 107u8
                                    && module_path[8usize] == 101u8
                                    && module_path[9usize] == 114u8
                                    && module_path[10usize] == 58u8
                                    && module_path[11usize] == 58u8
                                    && module_path[12usize] == 97u8
                                    && module_path[13usize] == 99u8
                                    && module_path[14usize] == 116u8
                                    && module_path[15usize] == 111u8
                                    && module_path[16usize] == 114u8
                                    && module_path[17usize] == 115u8
                                    && module_path[18usize] == 58u8
                                    && module_path[19usize] == 58u8
                                    && module_path[20usize] == 100u8
                                    && module_path[21usize] == 105u8
                                    && module_path[22usize] == 115u8
                                    && module_path[23usize] == 112u8
                                    && module_path[24usize] == 108u8
                                    && module_path[25usize] == 97u8
                                    && module_path[26usize] == 121u8
                                    && if 27usize == module_path.len() {
                                        true
                                    } else {
                                        module_path[27usize] == b':'
                                    }
                            } {
                                return true;
                            }
                            if if 10usize > module_path.len() {
                                false
                            } else {
                                module_path[0usize] == 100u8
                                    && module_path[1usize] == 105u8
                                    && module_path[2usize] == 115u8
                                    && module_path[3usize] == 99u8
                                    && module_path[4usize] == 111u8
                                    && module_path[5usize] == 98u8
                                    && module_path[6usize] == 105u8
                                    && module_path[7usize] == 107u8
                                    && module_path[8usize] == 101u8
                                    && module_path[9usize] == 114u8
                                    && if 10usize == module_path.len() {
                                        true
                                    } else {
                                        module_path[10usize] == b':'
                                    }
                            } {
                                return true;
                            }
                            false
                        }
                        check()
```
which ends up being 2.5KB. Those 2.5KB log calls add up, fast, and my final release binary ends up being 30% `check()` functions!

It looks like this code is being generated by `defmt_macros::function_like::log::env_filter::path_check`, which has the comment "Builds a compile-time check that returns `true` when `module_path!` can emit logs at the "requested log `level`". That may be true in optimized builds, but in a regular debug build this all gets compiled and executed at runtime.

jonas-schievink, 2022-08-15:
This should emit a const item, not a const fn, which would guarantee compile-time evaluation

Urhengulas, 2022-08-16:
@jonas-schievink said:
> This should emit a const item, not a const fn, which would guarantee compile-time evaluation

What do you think about just wrapping the current version in a `const CHECK: bool`? It does compile, but I don't know how to test if it actually solves the problem.

```rust
// current
if {
    const fn check() -> bool {
        if we_want_to_emit_log() {
            return true;
        }
        false
    }
    check()
} {
    emit_defmt_log();
}

// new
if {
    const CHECK: bool = {
        const fn check() -> bool {
            if we_want_to_emit_log() {
                return true;
            }
            false
        }
        check()
    };
    CHECK
} {
    emit_defmt_log();
}
```

I first tried to get rid of the function altogether, but this gives me an "error[E0572]: return statement outside of function body":
```rust
if {
    const CHECK: bool = {
        if we_want_to_emit_log() {
            return true;
        }
        false
    };
    CHECK
} {
    emit_defmt_log();
}
```
---
(All options are in this playground: https://play.rust-lang.org/?version=stable&mode=debug&edition=2021&gist=d82fe98724afa36aeb244d8d80c399ce)

Urhengulas, 2022-08-16:
It seems like it does not work. I adapted the `app-template` to use `defmt` as path dependency and collected data for the `levels` example, but there is no change in the code size.

That's the data:

| `--release` | size (current) | size (new) |
| :--- | :--- | :--- |
| false | 7984 | 7984 |
| true | 7364 | 7364 |

I collected the data like following:
```console
# current
$ DEFMT_LOG=trace cargo build --bin levels -q && size target/thumbv7em-none-eabihf/debug/levels
   text	   data	    bss	    dec	    hex	filename
   7984	     56	   1032	   9072	   2370	target/thumbv7em-none-eabihf/debug/levels

$ DEFMT_LOG=trace cargo build --bin levels -qr && size target/thumbv7em-none-eabihf/release/levels
   text	   data	    bss	    dec	    hex	filename
   7364	     56	   1032	   8452	   2104	target/thumbv7em-none-eabihf/release/levels

# new
$ DEFMT_LOG=trace cargo build --bin levels -q && size target/thumbv7em-none-eabihf/debug/levels
   text	   data	    bss	    dec	    hex	filename
   7984	     56	   1032	   9072	   2370	target/thumbv7em-none-eabihf/debug/levels

$ DEFMT_LOG=trace cargo build --bin levels -qr && size target/thumbv7em-none-eabihf/release/levels
   text	   data	    bss	    dec	    hex	filename
   7364	     56	   1032	   8452	   2104	target/thumbv7em-none-eabihf/release/levels

```

jonas-schievink, 2022-08-16:
The app-template uses `opt-level = 3` for debug builds, did you change that to 0 before testing?

Urhengulas, 2022-08-16:
> The app-template uses `opt-level = 3` for debug builds, did you change that to 0 before testing?

No, I did not. Will do that now.

Urhengulas, 2022-08-16:
Okay, now there is a difference:

| size (current) | size (new) | diff abs | diff % |
| :--- | :--- | :--- | :--- |
| 20728 | 16880 | -3848 | -18.6% |

commit a8b0fdf9 (2022-08-16): Merge #692

692: Wrap const fn in const item to ensure compile-time-evaluation. r=jonas-schievink a=Urhengulas

This reduces the size of the resulting binary in debug builds.

Fixes #691.
PR #692 (2022-08-16): Wrap const fn in const item to ensure compile-time-evaluation.
This reduces the size of the resulting binary in debug builds.

Fixes #691.
closed 2022-08-16 by bors[bot]