knurling-rs/defmt #693: Linker error on ESP32-C3: .text is not a multiple of alignment

TheButlah, 2022-08-27:
```
❯ cargo build
    Blocking waiting for file lock on package cache
    Blocking waiting for file lock on package cache
    Blocking waiting for file lock on package cache
   Compiling esp32c3-hal v0.1.0
   Compiling firmware v0.0.0 (/Users/ryan.butler/P/slimevr/rust/firmware)
warning: unused imports: `debug`, `info`
  --> src/main.rs:19:13
   |
19 | use defmt::{debug, error, info};
   |             ^^^^^         ^^^^
   |
   = note: `#[warn(unused_imports)]` on by default

error: linking with `rust-lld` failed: exit status: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "/var/folders/91/grpmpd313wj1v827k9005t6m0000gn/T/rustc3HGNWh/symbols.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.192lyajnjnn389a1.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.19lbkzw0vrd35kr.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.1anhacrvzeyteo99.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.1cpuulreqm9o00z5.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.1elm6t1du3s9m373.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.1eoa8tz51tlphfc9.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.1k7s585nxbde7bc9.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.1l2hteap4gzae274.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.1lj8fpa7g4stzkf8.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.1o10v5pz3wzpa3jp.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.1pjb27uroodhgg81.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.1rcop96is6waexfe.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.1rl7qop839lxxhsa.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.1uaualzoiskzbnbg.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.1vh8cmpbek6r5w4d.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.1wtvnwifzw453a7x.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.27gmv57j3dyi89m1.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.2dmzh2fw5nphjfg1.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.2poc697rqjwigbio.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.34dp148s3brbvcgp.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.3aknxefi4gfr51p8.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.3aqrn9l7kd0mn6u7.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.3c9mx4u0j0xj4y85.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.3lrclval8hb4nx57.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.3rb59b2j8wvg3419.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.3ukoua33w0hy7gl9.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.3us2eqjp52k132hb.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.3znb6f8h9lwycvb8.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.4ci26w5ld0lvkmf2.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.4vr29n1ltpidhuwu.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.5aqvawmnpg6yiick.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.5hqteunk4wlw6lk.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.6ayrszogpjoee1h.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.biuo4jbf2rn0wo1.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.bl3ezhb3yw0mi7v.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.g5o9sa9uf0agyjy.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.gembv8vqndopmax.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.h5qkjmv3caco9pt.rcgu.o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/firmware-4025a2a0ae46f9ca.1rzh4n5ls1no6om7.rcgu.o" "--as-needed" "-L" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps" "-L" "/Users/ryan.butler/P/slimevr/rust/firmware/target/debug/deps" "-L" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/build/defmt-a73b812ff69e04d5/out" "-L" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/build/riscv-3dc7557f74579c0d/out" "-L" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/build/esp32c3-hal-588e28f265a936a0/out" "-L" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/build/esp32c3-hal-588e28f265a936a0/out" "-L" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/build/esp32c3-08dd74e2ab6d1be1/out" "-L" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/build/riscv-rt-f476a4dd674c0474/out" "-L" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/build/riscv-atomic-emulation-trap-6f11bd41581605ae/out" "-L" "/Users/ryan.butler/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/riscv32imac-unknown-none-elf/lib" "--start-group" "-Bstatic" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/libesp_alloc-88c666d931228cd1.rlib" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/liblinked_list_allocator-f0bae4da17f7c074.rlib" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/libesp32c3_hal-916026dc0ff5afcd.rlib" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/libesp_hal_common-cf6d848da614b2c6.rlib" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/libriscv_atomic_emulation_trap-5053929a2da65e7b.rlib" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/libriscv_rt-ade39a2c8b42317e.rlib" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/libr0-66fd959a945244f1.rlib" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imac-unknown-none-elf/debug/deps/libfugit-7ecb672c2d19722b.rlib" "/Users/ryan.butler/P/slimevr/rust/firmwa [... cut ...]
runner = "probe-run --chip esp32c3"

[build]
rustflags = [
  # Required to obtain backtraces (e.g. when using the "esp-backtrace" crate.)
  # NOTE: May negatively impact performance of produced code
  "-C",
  "force-frame-pointers",

  "-C",
  "link-arg=-Tlinkall.x",

  # For defmt
  "-C",
  "link-arg=-Tdefmt.x",
]
target = "riscv32imac-unknown-none-elf"

[unstable]
build-std = ["core", "alloc"]
```

[Link to source code](https://github.com/SlimeVR/SlimeVR-Rust/tree/f277c2f0d0d39d56d7c167bd2ace93f872d98111/firmware)

Urhengulas, 2022-08-29:
This might be a risc-v specific error as well. Will do a bit of investigation.

Urhengulas, 2022-08-29:
It seems like we need to adapt our linker script to support risc-v targets, which we are not actively working on atm.

Fixing this likely requires to poke at our [linker script](https://github.com/knurling-rs/defmt/blob/a8b0fdf90b212c4845ca1f2be692b7f29a9179af/defmt/defmt.x.in). If you find a fix please share it here. Otherwise you'll unfortunately need to wait.

TheButlah, 2022-10-04:
This is fixed on `direct-boot` mode for the esp32c3, but not yet for the regular boot mode.

TheButlah, 2022-11-20:
This is *allegedly* fixed also on the esp-idf bootloader mode for the esp32c3, but I have not yet been able to confirm that it actually works. It definitely compiles.

TheButlah, 2022-11-22:
I have confirmed that this works now, both on direct-boot and on esp-bootloader.

closed 2022-11-22 by TheButlah