knurling-rs/defmt #694: Inconsistent Behaviour

sakian, 2022-08-29:
I've been using defmt over RTT for several months on an ATSAMD51J18A chip without issues. I've recently started developing for a ATSAMDA1G16B chip (64k FLASH, 8k RAM) and have been seen some inconsistent behaviour.

I have an example project that replicates the issue: https://github.com/sakian/atsamda1_test

Sometimes it works as expected:
```
    Finished dev [unoptimized + debuginfo] target(s) in 0.08s
     Running `probe-run --chip ATSAMDA1G16B target\thumbv6m-none-eabi\debug\atsamda1_test`
(HOST) INFO  flashing program (293 pages / 18.31 KiB)
(HOST) INFO  success!
────────────────────────────────────────────────────────────────────────────────
INFO  Started
└─ atsamda1_test::__cortex_m_rt_main @ src\main.rs:12
```

But sometimes I get a variant of a control block corrupted message like this:
```
    Finished dev [unoptimized + debuginfo] target(s) in 0.09s
     Running `probe-run --chip ATSAMDA1G16B target\thumbv6m-none-eabi\debug\atsamda1_test`
(HOST) INFO  flashing program (293 pages / 18.31 KiB)
(HOST) INFO  success!
────────────────────────────────────────────────────────────────────────────────
RTT error: Control block corrupted: write pointer is 4 while buffer size is 1 for "up" channel 0 ()
────────────────────────────────────────────────────────────────────────────────
Error: An error with the usage of the probe occurred

Caused by:
    Operation timed out
error: process didn't exit successfully: `probe-run --chip ATSAMDA1G16B target\thumbv6m-none-eabi\debug\atsamda1_test` (exit code: 1)
```
or this:
```
    Finished dev [unoptimized + debuginfo] target(s) in 0.09s
     Running `probe-run --chip ATSAMDA1G16B target\thumbv6m-none-eabi\debug\atsamda1_test`
(HOST) INFO  flashing program (293 pages / 18.31 KiB)
(HOST) INFO  success!
Error: Control block corrupted: Nonsensical array sizes at 20000000: max_up_channels=0 max_down_channels=15852
error: process didn't exit successfully: `probe-run --chip ATSAMDA1G16B target\thumbv6m-none-eabi\debug\atsamda1_test` (exit code: 1)
```

Things I've tried:
- Added support for the ATSAMDA1G16B target (was previously using ATSAMD21J16B)
- Reducing RTT buffer size to 256 and 64 bytes

Seems like I've configured things wrong, but can't figure it out. Does anyone have any ideas?

Also included a verbose output from probe-run:
```
    Finished dev [unoptimized + debuginfo] target(s) in 0.09s
     Running `probe-run --chip ATSAMDA1G16B --verbose target\thumbv6m-none-eabi\debug\atsamda1_test`
(HOST) DEBUG vector table: VectorTable { initial_stack_pointer: 20002000, hard_fault: 3be7 }
└─ probe_run::elf @ src\elf.rs:31
(HOST) DEBUG RAM region: 0x20000000-0x20001FFF
└─ probe_run::target_info @ src\target_info.rs:115
(HOST) DEBUG section `.data` is in RAM at 0x20000000..=0x2000002F
└─ probe_run::target_info @ src\target_info.rs:153
(HOST) DEBUG section `.bss` is in RAM at 0x20000030..=0x20000037
└─ probe_run::target_info @ src\target_info.rs:153
(HOST) DEBUG section `.uninit` is in RAM at 0x20000038..=0x20000437
└─ probe_run::target_info @ src\target_info.rs:153
(HOST) DEBUG valid SP range: 0x20000438..=0x20001FFC
└─ probe_run::target_info @ src\target_info.rs:167
(HOST) DEBUG found 1 probes
└─ probe_run::probe @ src\probe.rs:25
(HOST) DEBUG opened probe
└─ probe_run::probe @ src\probe.rs:33
(HOST) DEBUG started session
└─ probe_run @ src\main.rs:86
(HOST) INFO  flashing program (293 pages / 18.31 KiB)
└─ probe_run @ src\main.rs:98
(HOST) DEBUG Erased sector of size 4096 bytes in 84 ms
└─ probe_run @ src\main.rs:106
(HOST) DEBUG Erased sector of size 4096 bytes in 46 ms
└─ probe_run @ src\main.rs:106
(HOST) DEBUG Erased sector of size 4096 bytes in 45 ms
└─ probe_run @ src\main.rs:106
(HOST) DEBUG Erased sector of size 4096 bytes in 44 ms
└─ probe_run @ src\main.rs:106
(HOST) DEBUG Erased sector of size 4096 bytes in 45 ms
└─ probe_run @ src\main.rs:106
(HOST) DEBUG Programmed page of size 64 bytes in 46 ms
└─ probe_run @ src\main.rs:114
(HOST) DEBUG Programmed page of size 64 bytes in 31 ms
└─ probe_run @ src\main.rs:114
...
(HOST) DEBUG Programmed page of size 64 bytes in 35 ms
└─ probe_run @ src\main.rs:114
(HOST) DEBUG Programmed page of size 64 bytes in 31 ms
└─ probe_run @ src\main.rs:114
(HOST) DEBUG Programmed page of size 64 bytes in 35 ms
└─ probe_run @ src\main.rs:114
(HOST) INFO  success!
└─ probe_run @ src\main.rs:132
(HOST) DEBUG 7108 bytes of stack available (0x20000438 ..= 0x20001FFC), using 712 byte canary
└─ probe_run::canary @ src\canary.rs:87
(HOST) TRACE setting up canary took 0.040s (17.32 KiB/s)
└─ probe_run::canary @ src\canary.rs:101
(HOST) DEBUG starting device
└─ probe_run @ src\main.rs:190
(HOST) TRACE Could not attach because the target's RTT control block isn't initialized (yet). retrying
└─ probe_run @ src\main.rs:420
(HOST) DEBUG Successfully attached RTT
└─ probe_run @ src\main.rs:409
────────────────────────────────────────────────────────────────────────────────
RTT error: Control block corrupted: write pointer is 4 while buffer size is 0 for "up" channel 0 ()
────────────────────────────────────────────────────────────────────────────────
Error: An error with the usage of the probe occurred

Caused by:
    Operation timed out
error: process didn't exit successfully: `probe-run --chip ATSAMDA1G16B --verbose target\thumbv6m-none-eabi\debug\atsamda1_test` (exit code: 1)
```

sakian, 2022-08-30:
After more digging, this is a defmt issue. I removed defmt and depended only on RTT with probe-rs, and I'm seeing the same issues.

closed 2022-08-30 by sakian

Urhengulas, 2022-09-01:
> After more digging, this is a defmt issue. I removed defmt and depended only on RTT with probe-rs, and I'm seeing the same issues.

Do you mean that it is _not_ a defmt issue? Or that it _is_ a defmt issue?

sakian, 2022-09-01:
Yes, sorry, I meant to say it's not a defmt issue. `defmt-rtt` seems to take longer to initialize the RTT control block on my device, but I think it's an issue with the host side (`probe-run`).