knurling-rs/defmt #705: After upgrading to `defmt-rtt 0.4.0` building results in a `rust-lld: error`

eflukx, 2022-10-24:
Don't know this has really to do solely with defmt-rtt but when I upgraded the `defmt-rtt` in my project from 0.3.3 to 0.4.0 it building ends in ~tears~ a linker error. Platform/target is nRF9160, building a trivial hello world application with flip-link.

Actual  error here (https://gist.github.com/eflukx/1f797fcbb1e687d458754c4836eebd9a)

dpl0, 2022-10-25:
Hi @eflukx! 

I have also encountered this error before. It is happening because of the new dependency on `critical_section`'s last version. You have to use one of the implementations of it, just like in knurling's [app-template](https://github.com/knurling-rs/app-template/blob/main/Cargo.toml) in line 17 (this one is using one of the implementations for `cortex-m`, but I don't know how things work with nRF9160, honestly).

You could also check [critical_section's](https://docs.rs/critical-section/latest/critical_section/) documentation. Here it says that usually this is set up by HAL crates. Maybe your HAL crate is providing you with this feature, for instance.

eflukx, 2022-10-25:
Thanks! Actually the next thing I was planning to do was to re-iterate the Knurling template.. will have a look today!

I always find these linker errors so damned convoluted.. (or maybe that's just not have to deal with these very often.. luckyly :))

eflukx, 2022-10-25:
Yep, adding the `critical-section-single-core` to `cortex-m` fixed the issue for me! 👍🏻

closed 2022-10-25 by eflukx

Urhengulas, 2022-10-31:
Thank you for the report @eflukx, and glad that you could already solve it. We are aware of this breaking change, hence the new minor version and not just a patch. We added it to the `defmt-rtt` docs that one needs to add a `critical-section` implementation, but apparently, this was not enough. I would be interested in where you looked for but didn't find, the answer. So we can improve this for others.

eflukx, 2022-10-31:
@Urhengulas
Hi Johann, maybe this time I was a little 'trigger happy' to file this issue. It seemed so apparent to me this (link error) should not happen, therefore I might not have looked to deeply into the docs 😔... I expect developers to use the documentation more often as a guide when using defmt-rtt for the first time in a project. 

When I personally run into issues with some (open source) software component, I typically do a quick search though the Github issues first. More often then not I find people having the same/similar issue and often a solution to the problem is also presented. In my experience Github issues often give a quick handle on the things developers run into. Github issues (just like this one) can act as valuable addendum to the documentation (as they inherently incorporate the feedback from the developers using the lib/dependency/project).