knurling-rs/defmt #713: `defmt-rtt` `0.4` breaks on STM32H7A3ZIT6Q

indietyp, 2022-12-03:
After trying to upgrade, I encountered an issue with `defmt-rtt` `0.4`, no more messages will show up. I am unsure what the root cause is, and to verify that the cause isn't my code, I have created a [minimum reproducible example](https://github.com/indietyp/stlink-test) from the recommended app template. If I set `defmt-rtt = "0.4"` in the `Cargo.toml` I no longer get any output, while with `defmt-rtt = "0.3"` that isn't the case.

This is then executed - like in the README - on a [STM32H7A3 Nucleo-144](https://www.st.com/en/evaluation-tools/nucleo-h7a3zi-q.html).

rustc version: rustc 1.67.0-nightly (32e613bba 2022-12-02)

## `defmt-rtt = "0.4"`
[image]

## `defmt-rtt = "0.3"`
[image]

Urhengulas, 2022-12-06:
Can you please execute it with `-v` for `probe-run` and share the output?

indietyp, 2022-12-07:
Sure!

```bash
probe-run -V
# 0.3.5 
# supported defmt version: 3
```

## `defmt-rtt = "0.4"`

```log
(HOST) DEBUG vector table: VectorTable { initial_stack_pointer: 2001fbc0, hard_fault: 800170b }
└─ probe_run::elf @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/elf.rs:31
(HOST) DEBUG found 1 probes
└─ probe_run::probe @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/probe.rs:25
(HOST) DEBUG opened probe
└─ probe_run::probe @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/probe.rs:33
(HOST) DEBUG started session
└─ probe_run @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/main.rs:86
(HOST) INFO  flashing program (1 pages / 32.00 KiB)
└─ probe_run @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/main.rs:98
(HOST) DEBUG Erased sector of size 8192 bytes in 30 ms
└─ probe_run @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/main.rs:106
(HOST) DEBUG Erased sector of size 8192 bytes in 13 ms
└─ probe_run @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/main.rs:106
(HOST) DEBUG Erased sector of size 8192 bytes in 12 ms
└─ probe_run @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/main.rs:106
(HOST) DEBUG Erased sector of size 8192 bytes in 14 ms
└─ probe_run @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/main.rs:106
(HOST) DEBUG Programmed page of size 32768 bytes in 84 ms
└─ probe_run @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/main.rs:114
(HOST) INFO  success!
└─ probe_run @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/main.rs:133
(HOST) DEBUG couldn't find valid stack range, not placing stack canary
└─ probe_run::canary @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/canary.rs:64
(HOST) DEBUG starting device
└─ probe_run @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/main.rs:191
(HOST) DEBUG Successfully attached RTT
└─ probe_run @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/main.rs:410
────────────────────────────────────────────────────────────────────────────────
~────────────────────────────────────────────────────────────────────────────────
(HOST) TRACE 0x0800040a: found FDE for 0x0800040a .. 0x0800040e at offset 10960
└─ probe_run::backtrace::unwind @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/backtrace/unwind.rs:231
(HOST) TRACE uwt row for pc 0x0800040a: UnwindTableRow { start_address: 134218762, end_address: 134218766, saved_args_size: 0, cfa: RegisterAndOffset { register: Register(13), offset: 0 }, registers: RegisterRuleMap { rules: [] } }
└─ probe_run::backtrace::unwind @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/backtrace/unwind.rs:78
(HOST) DEBUG LR=0x08000349 PC=0x0800040A
└─ probe_run::backtrace::unwind @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/backtrace/unwind.rs:88
(HOST) TRACE 0x08000348: found FDE for 0x08000340 .. 0x0800034a at offset 124
└─ probe_run::backtrace::unwind @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/backtrace/unwind.rs:231
(HOST) TRACE uwt row for pc 0x08000348: UnwindTableRow { start_address: 134218564, end_address: 134218570, saved_args_size: 0, cfa: RegisterAndOffset { register: Register(7), offset: 8 }, registers: RegisterRuleMap { rules: [(Register(14), Offset(-4)), (Register(7), Offset(-8))] } }
└─ probe_run::backtrace::unwind @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/backtrace/unwind.rs:78
(HOST) DEBUG update_cfa: CFA changed Some(2001fb9c) -> 2001fba4
└─ probe_run::registers @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/registers.rs:52
(HOST) TRACE update reg=Register(14), rule=Offset(-4), abs=0x2001fba0 -> value=0x08000339
└─ probe_run::registers @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/registers.rs:72
(HOST) TRACE update reg=Register(7), rule=Offset(-8), abs=0x2001fb9c -> value=0x2001fbac
└─ probe_run::registers @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/registers.rs:72
(HOST) DEBUG LR=0x08000339 PC=0x08000348
└─ probe_run::backtrace::unwind @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/backtrace/unwind.rs:88
(HOST) TRACE 0x08000338: found FDE for 0x08000318 .. 0x08000340 at offset 48
└─ probe_run::backtrace::unwind @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/backtrace/unwind.rs:231
(HOST) TRACE uwt row for pc 0x08000338: UnwindTableRow { start_address: 134218524, end_address: 134218560, saved_args_size: 0, cfa: RegisterAndOffset { register: Register(7), offset: 8 }, registers: RegisterRuleMap { rules: [(Register(14), Offset(-4)), (Register(7), Offset(-8))] } }
└─ probe_run::backtrace::unwind @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/backtrace/unwind.rs:78
(HOST) DEBUG update_cfa: CFA changed Some(2001fba4) -> 2001fbb4
[... cut ...]
└─ probe_run::backtrace::symbolicate @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/backtrace/symbolicate.rs:107
(HOST) TRACE demangle Ok("main") (language=Some(DwLang(1C))) -> Ok("main")
└─ probe_run::backtrace::symbolicate @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/backtrace/symbolicate.rs:107
(HOST) INFO  device halted without error
└─ probe_run::backtrace @ /Users/bmahmoud/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/backtrace/mod.rs:104
```

idaniel86, 2022-12-28:
Hi, I am having the same issue with STM32F3DISCOVERY board. Try tried examples provided in the template bin folder. When switching back to version 0.3 it works.

Urhengulas, 2023-01-10:
I just tried reproducing it with a `STM32F429ZITx`, but it works for me with both `0.3` and `0.4`.

Urhengulas, 2023-01-10:
I was finally able to reproduce the bug with a STM32F3DISCOVERY board 🎉 

After bisecting, I found that #683 is the problem. @jannic Do you know why this problem might be caused by your change?

jannic, 2023-01-10:
@Urhengulas thanks for that great detective work!
Unfortunately, I have no idea how #683 could cause this.

Looking at the debug output and the probe-run source code, I'd say this means that probe-run did find the RTT block in RAM (it found the SEGGER_RTT signature), but then the check in https://github.com/knurling-rs/probe-run/blob/v0.3.5/src/main.rs#L260 doesn't find the expected channel name `defmt`.

But that's really strange, as the data section looks correct:
```
Contents of section .data: 
 2001fbc0 53454747 45522052 54540000 00000000  SEGGER RTT......
 2001fbd0 01000000 00000000 f0fb0120 00fc0120  ........... ...
 2001fbe0 00040000 00000000 00000000 01000000  ................
 2001fbf0 6465666d 74000000                    defmt...
```

```
2001fbc0 D __sdata
2001fbc0 D _SEGGER_RTT
2001fbf0 d defmt_rtt::handle::NAME::h8a4d067f21f224a5
```
The RTT block is as the start of the data section, at 2001fbd0 there is the pointer to the name, and at 2001fbf0 there is the expected value.

In theory, it could be a race condition: The debugger could read that RAM range after the RTT data structure was initialized, but before the `defmt` name was written. But, without knowing details about the initialization code, this seems to be a very small time window the debugger would need to hit.

Could you add some debug output to verify the theory, that the name read doesn't match the expected value `defmt`? What does it read instead?
Does it help to add some delay to probe-run, before it reads the RTT structures? For example at the beginning of `setup_logging_channel`? But to be honest I don't think this explanation holds water. If this was a race condition, how could it reliably be triggered, given that the readout happens over a relatively slow USB interface? There should be plenty of jitter, so the exact timing when the RTT structure is read just can't be deterministic.

Does this chip use some unusual RAM initialization?

jannic, 2023-01-10:
I found a likely cause:

The code reading the name is one of the few places where an address is checked if it is actually part of some valid memory mapping: https://github.com/probe-rs/probe-rs/blob/master/rtt/src/channel.rs#L369-L373
In contrast, the RTT channel header itself is read without such a check. So if the memory regions in the chip definition used by probe-run are wrong, one would observe exactly the effect seen by @indietyp.

But why would the memory regions be wrong? Well, that may be caused by https://github.com/probe-rs/probe-rs/issues/429
That issue was fixed in October, but that fix isn't part of any release yet.

@Urhengulas, when you reproduced the error with an STM32F3DISCOVERY, was the RAM size specified in memory.x consistent with the target chip definition used by probe-run?

jannic, 2023-01-13:
A new version, 0.14.0, of probe-rs was just released. It contains the fix for https://github.com/probe-rs/probe-rs/issues/429.

Urhengulas, 2023-01-25:
> A new version, 0.14.0, of probe-rs was just released. It contains the fix for [probe-rs/probe-rs#429](https://github.com/probe-rs/probe-rs/issues/429).

@indietyp Can you please test this with `probe-run v0.3.6` which includes `probe-rs v0.14`? Just run `cargo install probe-run` to upgrade.

I've just tried it with `probe-rs v0.14` and the STM32F3, but this, unfortunately still didn't work. But the issue jannic mentioned is only about the STM32H7, so it was not to be expected to change anything for the STM32F3.

Urhengulas, 2023-01-26:
:tada: The logs appear on the `STM32F303VCT6` when applying the following patch, which makes `probe-rs` use the `RAM` instead of the `CCMRAM`:
```diff
diff --git a/probe-rs/targets/STM32F3_Series.yaml b/probe-rs/targets/STM32F3_Series.yaml
index ec040c59..c480f06f 100644
--- a/probe-rs/targets/STM32F3_Series.yaml
+++ b/probe-rs/targets/STM32F3_Series.yaml
@@ -1181,8 +1181,8 @@ variants:
     memory_map:
       - !Ram
           range:
-            start: 0x10000000
-            end: 0x10002000
+            start: 0x20000000
+            end: 0x2000a000
           is_boot_memory: false
           cores:
             - main
``` 

```console
$ cargo rb hello
    Finished dev [optimized + debuginfo] target(s) in 0.03s
     Running `probe-run --chip STM32F303VCTx target/thumbv7em-none-eabihf/debug/hello`
(HOST) INFO  flashing program (8 pages / 8.00 KiB)
(HOST) INFO  success!
────────────────────────────────────────────────────────────────────────────────
Hello, world!
└─ hello::__cortex_m_rt_main @ src/bin/hello.rs:8
────────────────────────────────────────────────────────────────────────────────
(HOST) INFO  device halted without error
```

There are two options to move forward (in order of preference):
- Teach `probe-rs` about the second memory region. This depends on `probe-rs`' capabilities regarding multi-RAM chips.
- Teach `defmt-rtt` to place the RTT header into `CCRAM` instead of `RAM`
I will investigate both.

Urhengulas, 2023-02-05:
The issue was fixed for the `STM32H7` by https://github.com/probe-rs/probe-rs/pull/1268, which is already released. The issue is also fixed for the `STM32F3` by https://github.com/probe-rs/probe-rs/pull/1436, which is not released yet.

closed 2023-02-05 by Urhengulas

stonefoxes, 2023-02-15:
Issue still exists on STM32L451CCU6 with actual versions from cargo (found on knurling app template).

Urhengulas, 2023-02-15:
@stonefoxes said:
> Issue still exists on STM32L451CCU6 with actual versions from cargo (found on knurling app template).

Can you please raise that in the `probe-rs` repo? You can probably also draft a PR similar to https://github.com/probe-rs/probe-rs/pull/1436 and then it is fixed with the next probe-rs release.