knurling-rs/defmt #723: `Format::format` causes HardFault

matoushybl, 2023-01-13:
Hi!
I have a manual implementation of defmt::Format for an error type like this:
```rust
#[derive(Debug)]
#[cfg_attr(feature = "defmt", derive(defmt::Format))]
pub enum ParsingError {
    Crc,
}

pub enum Error<Inner: core::fmt::Debug> {
    Bus(Inner),
    Parsing(ParsingError),
}

#[cfg(feature = "defmt")]
impl<E: embedded_hal_async::i2c::Error + defmt::Format> defmt::Format for Error<E> {
    fn format(&self, fmt: defmt::Formatter) {
        match self {
            Error::Bus(e) => e.format(fmt),
            Error::Parsing(e) => e.format(fmt),
        }
    }
}
```

When I try to run the following code, both of the error macros crash with hardfault.
```rust
    defmt::error!(
        "err: {}",
        Error::Bus(embassy_nrf::twim::Error::DataNack)
    );
    defmt::error!(
        "err: {}",
        Error::<embassy_nrf::twim::Error>::Parsing(
            sensirion_async::ParsingError::Crc
        )
    );
```

```
stack backtrace:
   0: HardFaultTrampoline
      <exception entry>
   1: <defmt_rtt::Logger as defmt::traits::Logger>::acquire
   2: _defmt_acquire
        at /Users/matous/.cargo/registry/src/github.com-1ecc6299db9ec823/defmt-rtt-0.4.0/src/lib.rs:43:1
   3: <sensirion_async::ParsingError as defmt::traits::Format>::format
        at /Users/matous/Development/Rust/Embedded/Projects/air-force-one/sensirion-async/src/lib.rs:40:38
   4: core::sync::atomic::atomic_load
        at /Users/matous/.rustup/toolchains/nightly-2022-11-22-aarch64-apple-darwin/lib/rustlib/src/rust/library/core/src/sync/atomic.rs:2987:24
   5: core::sync::atomic::AtomicPtr<T>::load
        at /Users/matous/.rustup/toolchains/nightly-2022-11-22-aarch64-apple-darwin/lib/rustlib/src/rust/library/core/src/sync/atomic.rs:1191:18
   6: SPIM0_SPIS0_TWIM0_TWIS0_SPI0_TWI0
        at src/bin/main.rs:92:15
```

When I implement the `defmt::Format` trait using the derive macro, the HardFault doesn't happen. Am I doing something wrong? I will use the derive macro, but I am wondering what could cause a crash like this.

Thanks!

Urhengulas, 2023-01-23:
Hello @matoushybl,

The problem is that `defmt` doesn't allow forwarding the formatting the way you do. As written in our [user docs](https://defmt.ferrous-systems.com/format.html#manual-implementation-with-write) you can achieve your goal like this:
```rust
fn format(&self, fmt: defmt::Formatter) {
    match self {
        Error::Bus(e) => defmt::write!(fmt, "{}", e),
        Error::Parsing(e) => defmt::write!(fmt, "{}", e),
    }
}
```

Have fun using defmt!

closed 2023-01-23 by Urhengulas

matoushybl, 2023-02-20:
Hi,
Thanks for the reply! However, should a mistake like this result in a HardFault rather than a compile-time or a runtime error?
Thanks!

Urhengulas, 2023-02-20:
> Hi, Thanks for the reply! However, should a mistake like this result in a HardFault rather than a compile-time or a runtime error? Thanks!

That is true, but I don't know if we are able to detect this somehow.