knurling-rs/defmt #730: Adding a std logging backend

diondokter, 2023-01-27:
So I've been using defmt in a couple of libraries of mine and it's super nice!

However, when I want to add tests to the library that I run on my PC, defmt needs a global logger and there are no loggers that work for PC tests afaik.
What I want to propose is similar to #714, but maybe not quite the same.

I would like to see a feature flag that switches defmt over from using its own logger and deferred formatting to using println, panic, the log crate and Rust's built-in formatter.

I took a stab at it today, but only for the `defmt::println` macro to try this out: https://github.com/knurling-rs/defmt/compare/main...diondokter:defmt:std-log

In essence I think it should be possible to translate the defmt formatting to the rust formatting which can then be routed to `panic`, `println` and `log::****` as required.

This is pretty doable if the API would be that every type needs to implement Debug when in this mode, but theoretically it should be possible to use the `Format` trait as input too.

So I want to know, is there interest in this? Has something in this direction a chance of being accepted?
If so I think I could spare some time to work on this.

Urhengulas, 2023-01-27:
Hi @diondokter, I would like this. Many people report that they struggle with this issue, especially in CI. Since it would be behind a feature flag it is not even a breaking change as far as I understand.

Also, we mostly tried to stay close to the `std::fmt` grammar for the format strings, so it _should_ be possible to translate with not too much effort. Please tell me about the cases where the `defmt` grammar diverges from the `std::fmt` grammar.

diondokter, 2023-01-28:
Based on the defmt book I can see:

- Primitive specifications. But these can mostly be ignored, except for the bitfields and interned strings (https://defmt.ferrous-systems.com/primitives.html)
- The `:a` ascii hint
- The `:us` hint

t-moe, 2024-11-02:
can you comment on whether this is on the v1.0.0 roadmap @jonathanpallant ?
I took another stab at this today to try how we can modify the existing `derive(Format)`, so that it also works with the log backend. ...Its kinda tricky without introducing semver-breaking changes. So it would be nice to know, what is planned regarding this topic...

jonathanpallant, 2024-11-04:
Working on it!

jonathanpallant, 2024-11-05:
So, my current thinking is to merely stabilise the syntax of the `defmt::info!` macros (et al) for 1.0.

After that, I think there are two options for `defmt` work on libstd systems:

1. Add a feature flag for *not* interning strings, so the string is serialised with the rest of the values. You can then write a `defmt-log` crate which parses the format string, renders it to a `String` and throws it at e.g. `log::info!`.
   * This requires all arguments to implement defmt::Format  
2. Add a feature flag for *using core::fmt*. That would make `defmt` process the format strings with a proc-macro to remove the `=u32` style type specifiers, and hand the modified format string onto the `log::info!` macro. 
   * This requires all arguments to implement `defmt::Format` *and* `core::fmt::Debug`

There might be other options as well, but I think we can make a decision about which approach is best *after* 1.0 is out.

t-moe, 2024-11-05:
Hmm, I hadn’t considered option 1. But you’re right; since the std-logging backend will likely be used only on powerful (non-embedded) targets, using defmt encoding, followed by decoding and forwarding to log, could indeed be feasible.

The approach I tested locally is similar to your Option 2, but without requiring all arguments to implement `core::fmt::Debug`.

Instead I added a new hidden method to the `Format` trait, which basically mirrors `core::fmt::Debug::fmt()`:
```rust
pub trait Format {
  //...

  #[doc(hidden)]
  fn _core_fmt(&self, fmt: &mut fmt::Formatter<'_>) -> fmt::Result {
    Ok(())  
  }
}
```

Then I've extended the codegen logic for all defmt macros (e.g. `defmt::error!`) to wrap each argument with `Format2Debug(&arg)`, where `Format2Debug` is defined as:

```rust
pub struct Format2Debug<'a, T: Format + ?Sized>(pub &'a T);
impl <'a, T: Format + ?Sized> fmt::Debug for Format2Debug<'a, T> {
    fn fmt(&self, fmt: &mut fmt::Formatter<'_>) -> fmt::Result {
     self.0._core_fmt(fmt)
    }
}
```

Next I've extended the `derive(defmt::Format)` implementation to automatically populate the body of this method. And I've also implemented the `_core_fmt` method for all primitives etc.

 
The advantage of this solution is, that as a user of defmt I dont need to implement `core::fmt::Debug` and can keep using `defmt::Format`  in my code and enable the std logging backend with a simple feature flag. Also, its not too much effort to implement this whole change. (Probably less than implementing `defmt-log` ?)

The problem arises, and this is where I need your attention, if the user manually implements `defmt::Format`:
I think it would be best if we could pack the `&mut core::fmt::Formatter<'_>` into our `defmt::Formatter`, so that the `write!` calls in the body of the `defmt::Format::fmt` implementations can forward to the `core::fmt::Formatter`.  But this doesnt work currently, because `defmt::Formatter` is passed by value and implements `Copy`.  Therefore, if we ever want to follow this approach, I propose applying the following change before 1.0.0:

```diff
pub trait Format {
-  fn format(&self, fmt: Formatter);
+  fn format(&self, fmt: &mut Formatter<'_>);

 //...
}
```

Urhengulas, 2024-12-02:
There is previous discussion in https://github.com/knurling-rs/defmt/issues/463.

t-moe, 2024-12-04:
I’d like to renew my request to consider including the highlighted diff above into defmt 1.0.

In [this comment](https://github.com/knurling-rs/defmt/issues/463#issuecomment-2511794495), @Dirbaio mentioned:

> (fwiw https://github.com/knurling-rs/defmt/issues/730 can be solved at the application layer by making it use either log or defmt. embassy uses [this](https://github.com/embassy-rs/embassy/blob/main/embassy-net/src/fmt.rs) for example. Getting actual tokenization working on non-embedded does require fixes in defmt)

While this workaround is functional, it introduces a frustrating user experience:

* Every project must either copy `fmt.rs` or rely on something like `defmt-or-log` (which ideally wouldn’t need to exist).
* Expanded file and line numbers always point to `fmt.rs` instead of the actual source location.
* Both `defmt::Format` and `core::fmt::Debug` traits need to be implemented in the application, often requiring `#[cfg_attr]` and other boilerplate.

I think it should be just a matter of enabling a feature on the defmt crate and having it work on non-embedded targets.

defmt has become the de facto standard for embedded Rust, yet we still encounter unnecessary boilerplate and log/defmt switching in every new project. I don’t have a strong preference for whether #730 or #463 is implemented first, but addressing either would significantly improve the developer experience.

Dirbaio, 2024-12-04:
> Expanded file and line numbers always point to fmt.rs instead of the actual source location.

this is no longer the case with `#[collapse_debuginfo(yes)]`

t-moe, 2024-12-04:
> this is no longer the case with `#[collapse_debuginfo(yes)]`

Interesting, TIL.

Still, there is the problem that we have no common trait. How do i write a function which accepts a struct that can be formatted with either log or demft (depending on whether I'm compiling for CI/CD tests or embedded,   log or defmt)

pvdrz, 2025-02-08:
I think it is possible to change `defmt::Format::format` so it takes `&mut Formatter<'_>` instead of just `Formatter` and add a small compatibility layer for folks that already implement `Format` manually.

This layer would be behind a feature that I'm going to call `compat` (naming things is hard™). This feature would be enabled by default for `defmt` on `0.3.x` and disabled by default on `1.0.0` and it would gate a slightly tweaked version of the `Format` trait:

```rust
pub trait Format<F = Formatter<'static>> where F: CompatFormatter {
   fn format(&self, fmt: F);
   // ...
}
```

Here `CompatFormatter` would basically be a sealed trait that's only implemented for `Formatter<'a>` and for `&'a mut Formatter<'b>`. This means that any `impl Format for T { ... }` block would not need any changes when the `compat` feature is enabled. 

If users want to move forward, they need to do the following change:
```diff
- impl Format for T {
+ impl<F: CompatFormat> Format<F> for T {
-    fn format(&self, fmt: Formatter);
+    fn format(&self, fmt: F);  
    // ...
}
```

Which means their implementations could use either `Formatter` or `&mut Formatter<'_>` without issue. Or they could do the change to `&mut Formatter<'_>` directly and disable the `compat` flag all together.

Let me know what you think!

t-moe, 2025-02-10:
Hey @pvdrz,

Thanks for considering a backward-compatible approach for releasing this change — it does seem like a viable solution.

That said, I believe this is more of a strategic decision than a purely technical one. Should we include this change in 1.0 to keep the door open for future developments, or is it unnecessary because we're likely to take a different path (like #890)?

I'm not sure who has the authority to decide this — perhaps the Embedded Rust WG or someone else?

jonathanpallant, 2025-02-10:
For reference, Ferrous Systems hold the decision making responsibility, as the owners of the knurling-rs project.

jonathanpallant, 2025-02-12:
@t-moe - is your defmt fork hosted somewhere so I can see how you've done your implementation?

t-moe, 2025-02-12:
I havent look at it in a while, but this was my working branch back then:
https://github.com/knurling-rs/defmt/compare/main...t-moe:defmt:std_logging

t-moe, 2025-03-01:
@jonathanpallant  In which direction are we going with this?

Would you take a PR that changes that signature (before defmt 1.0) ^^, and subsequently implement streaming to the std logging backend (as outlined above)? 
Or do you want to take a different approach and/or will tackle this internally at Ferrous Systems ?

jonathanpallant, 2025-03-03:
I need to find the time to go through this in detail to satisfy myself that the breaking change (and the clunky workaround we have for 0.3 users) is absolutely necessary. I'll try and get it done this week.

chrysn, 2025-03-03:
I don't have the full picture, but might it be part of the solution to "just" leave the text strings in the binary if some tool asks it (where now rather than being a single u8 with an ignored value, the symbol points to the actual UTF-8 slice), or to have a parallel array (with a global defmt-"private" name) that uses the same index space but contains the slices? That would not alter the produced binaries unless a consumer such as a defmt-to-log explicitly enables the feature to emit these.

jonathanpallant, 2025-03-04:
OK, so I've spent some time looking into this.

The approach here is to add secret `_core_fmt` method to the `defmt::Format` trait, which means that anything that implements `defmt::Format` can be sent to the standard `write!` mechanism by means of an adapter type `Format2Debug` which accepts any type that implements `defmt::Format` and implements `core::fmt::Debug`.

This approach fundamentally means that the rendering is performed in two different ways - once through the `Format::format` method and once through the `Format::_core_fmt` method - and these functions may produce different results (or at least we cannot constrain them to the do the same thing). There's then the difficult of automatically implementing the method, especially when people `impl defmt::Format for MyType` themselves.

I propose an alternative approach. It is less performant, but more consistent.

By adding a change to defmt to optionally *not* intern strings, and to use the address of the `&'static str` (so a `*constant &'static str`) as the string ID, you can make it so that the defmt transport has full access to all information - the byte stream contains the address of the interned format strings, the address of any other interned strings, and any other parameters to be passed to the format strings. All that then remains (ha!) is to write a defmt-log transport crate which parses the byte stream, and formats the output according to the format string and the byte sequence.

I have a work-in-progress which can do this:

```console
[2025-03-03T15:27:59Z INFO  defmt_log] Acquiring ThreadId(1)
[2025-03-03T15:27:59Z DEBUG defmt_log] Bytes arrived on ThreadId(1): [00, c0, 93, 00, 01, 00, 00, 00]
[2025-03-03T15:27:59Z INFO  defmt_log] Format string: {"package":"desktop-demo","tag":"defmt_info","data":"Logging at info. u32 = {}","disambiguator":"11240794377383153101","crate_name":"desktop_demo"}
[2025-03-03T15:27:59Z DEBUG defmt_log] Bytes arrived on ThreadId(1): [20, c0, 93, 00, 01, 00, 00, 00]
[2025-03-03T15:27:59Z INFO  defmt_log] Arg: [20, c0, 93, 00, 01, 00, 00, 00]
[2025-03-03T15:27:59Z DEBUG defmt_log] Bytes arrived on ThreadId(1): [64, 00, 00, 00]
[2025-03-03T15:27:59Z INFO  defmt_log] Arg: [64, 00, 00, 00]
[2025-03-03T15:27:59Z INFO  defmt_log] Releasing ThreadId(1)
[2025-03-03T15:27:59Z INFO  defmt_log] "Logging at info. u32 = XXX" (package="desktop-demo", crate_name="desktop_demo", tag="defmt_info")
[2025-03-03T15:27:59Z INFO  defmt_log] Acquiring ThreadId(1)
[2025-03-03T15:27:59Z DEBUG defmt_log] Bytes arrived on ThreadId(1): [10, c0, 93, 00, 01, 00, 00, 00]
[2025-03-03T15:27:59Z INFO  defmt_log] Format string: {"package":"desktop-demo","tag":"defmt_warn","data":"Logging at warn. str = {}","disambiguator":"10286149126245529780","crate_name":"desktop_demo"}
[2025-03-03T15:27:59Z DEBUG defmt_log] Bytes arrived on ThreadId(1): [30, c0, 93, 00, 01, 00, 00, 00]
[2025-03-03T15:27:59Z INFO  defmt_log] Arg: [30, c0, 93, 00, 01, 00, 00, 00]
[2025-03-03T15:27:59Z DEBUG defmt_log] Bytes arrived on ThreadId(1): [08, 00, 00, 00]
[2025-03-03T15:27:59Z INFO  defmt_log] Arg: [08, 00, 00, 00]
[2025-03-03T15:27:59Z DEBUG defmt_log] Bytes arrived on ThreadId(1): [41, 20, 73, 74, 72, 69, 6e, 67]
[2025-03-03T15:27:59Z INFO  defmt_log] Arg: [41, 20, 73, 74, 72, 69, 6e, 67]
[2025-03-03T15:27:59Z INFO  defmt_log] Releasing ThreadId(1)
[2025-03-03T15:27:59Z WARN  defmt_log] "Logging at warn. str = XXX" (package="desktop-demo", crate_name="desktop_demo", tag="defmt_warn")
```

The `log` crate output shows that I have recovered the format strings and I can walk through it producing literal string segments and parameter segments. What I cannot do currently is to pull the parameters from the byte stream. The defmt-decoder crate *can* do this, but currently it assumes you always have an ELF file, and I do not. So, the next step is to break apart the defmt-decoder crate so that it is possible to format any arbitrary defmt frame using some 'String Fetcher', which in probe-rs is going to scan the ELF, but for defmt-print might also be able to scan some JSON dump of the string table, and for defmt-log will just cast the number to a `*const &'static str` and retrieve the string.

There is some cost to deserialising the byte sequence, but the upside is that the output will look exactly like it does in probe-rs, with the same formatting and the same `x=u8:a` annotations all working exactly the same. I believe that if you're sending logs to the `log` crate but using `defmt` for logging, you are probably running tests, or at least not that interested in whether a log message takes nanoseconds or microseconds.

jonathanpallant, 2025-03-04:
> I don't have the full picture, but might it be part of the solution to "just" leave the text strings in the binary if some tool asks it (where now rather than being a single u8 with an ignored [...]

This is my intention, behind a feature called 'no-intern' or something. Putting them in an array is non-trivial, because you have lots of crates with lots of interned strings and the linker needs to put them together, and linkers can't put them in arrays. Linkers *can* put things into a section, and if you have a section full of nothing but objects of the same size, you can treat them like an array, but it requires linker script shenanigans that don't really work on Desktop OSes (where people never supply custom linker scripts and/or the linker isn't GNU ld compatible).

t-moe, 2025-10-30:
Hey @jonathanpallant,
is there any update on this topic? I recall you were working on a prototype.

Also, is there a roadmap you could share?
I’d love to see defmt receive regular updates and continued development. I understand that this takes both time and funding — I might be in a position to help enable that, for example by supporting engineering work on the project.

Feel free to reach out on Matrix if you’d like to discuss this further.

t-moe, 2025-11-14:
@skade @jonathanpallant 
Thanks for your feedback on my previous comment and for taking the time to respond via GitHub/Matrix! 

I won’t be able to start working on this until January, as I’m planning to hire additional talent for my company who can then contribute to this feature.

I also wanted to point out a newly created issue in the Embedded Rust WG regarding a broader funding initiative for Embedded Rust projects: [#884](https://github.com/rust-embedded/wg/issues/884). Regardless of whether that initiative moves forward or not, I still plan to continue contributing to defmt directly.

elpiel, 2026-01-30:
> Not sure if this is the correct place to post this, but I created [defmt-stdout](https://crates.io/crates/defmt-stdout) based on code in #463, to make defmt usable on Linux

That's great! Code looks straight forward, I might give it a try over the weekend.