knurling-rs/defmt #731: defmt-test is not ignoring tests correctly

liam-b, 2023-02-01:
It seems to know that the test should be ignored but runs it anyway. I have a [minimal reproducible example repo](https://github.com/liam-b/defmt-test-ignore-bug) based on the app template modified for the [Pixhawk 6C](http://www.holybro.com/product/pixhawk-6c/) (running an STM32H743).

I did have to move `defmt-rtt` to version 0.3 because of #713 and I couldn't get `flip-link` working yet so it is disabled.

Running `cargo test --test integration` gives me:
```
   Compiling defmt-test-ignore-bug v0.1.0 (/Users/liam/src/tmp/defmt-test-ignore-bug)
    Finished test [optimized + debuginfo] target(s) in 0.71s
     Running tests/integration.rs (target/thumbv7em-none-eabihf/debug/deps/integration-7afe1656c338520d)
(HOST) INFO  flashing program (8 pages / 8.00 KiB)
(HOST) INFO  success!
────────────────────────────────────────────────────────────────────────────────
(1/2) running `it_works`...
└─ integration::tests::__defmt_test_entry @ tests/integration.rs:13
(2/2) ignoring `it_does_not_work`...
└─ integration::tests::__defmt_test_entry @ tests/integration.rs:19
ERROR panicked at 'should be ignored'
└─ integration::tests::it_does_not_work @ tests/integration.rs:20
────────────────────────────────────────────────────────────────────────────────
(HOST) WARN  no RAM region appears to contain the stack; probe-run can't determine if this was a stack overflow
stack backtrace:
   0: HardFaultTrampoline
      <exception entry>
   1: lib::inline::__udf
        at ./asm/inline.rs:181:5
   2: __udf
        at ./asm/lib.rs:51:17
   3: cortex_m::asm::udf
        at /Users/liam/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.7.7/src/asm.rs:43:5
   4: _defmt_panic
        at src/lib.rs:14:5
   5: main
   6: Reset
(HOST) ERROR the program panicked
error: test failed, to rerun pass `--test integration`

Caused by:
  process didn't exit successfully: `probe-run --chip STM32H743VIHx /Users/liam/src/tmp/defmt-test-ignore-bug/target/thumbv7em-none-eabihf/debug/deps/integration-7afe1656c338520d` (exit status: 6)
```

Urhengulas, 2023-02-03:
Hi @liam-b, `defmt-test` doesn't support a `#[ignore]` attribute at the moment. If you implement it, I am happy to review and merge it.

jonas-schievink, 2023-03-25:
#618 implemented support for `#[ignore]`, but has a bug where the function will be run regardless: https://github.com/knurling-rs/defmt/pull/618#discussion_r1148160200

Urhengulas, 2024-12-02:
A call to the ignored function is currently added here: https://github.com/knurling-rs/defmt/blob/main/firmware/defmt-test/macros/src/lib.rs#L372-L373.

Instead there should be no call, but we should print that the test is ignored.

anirudh12032008, 2026-11-12:
Picking this up. The ignored test still runs because the macro emits `let _ = #call;` for `#[ignore]`. I'll drop the call (keeping a plain reference to the fn so there's no dead_code warning) and make the qemu `ignored()` test panic so the snapshot proves it's skipped.