knurling-rs/defmt #732: impl Format for core::fmt::Arguments

elpiel, 2023-02-09:
I was using defmt today an I realised that there isn't an implementation for `core::fmt::Arguments`.
Is there a reason that it cannot be implemented?
I'm also interested in working on this.

Urhengulas, 2023-02-11:
I don't see why we should not have a `Format` implementation for it. If I understand the docs correctly it will be evaluated during compile-time and not use `core::fmt` code on the device.

But I am wondering what is your usecase?

elpiel, 2023-02-12:
After publishing the issue I realised that Arguments cannot be stand-alone, still not sure why though...

I was looking for a way to use `format_args!` to create a single message to be used across `defmt` for logging and writing to a log file because right now I need to put it in something like `heapless::String` to remove the duplication.

Maybe implementing Format for Arguments is not needed as I it can't be used alone to be put in a variable?!

Urhengulas, 2023-02-15:
I am not sure if I understand @elpiel. Could you give an example of what you are trying to achieve?

elpiel, 2023-03-11:
Using formatting that's not supported by defmt, like floating point numbers with given precision `info!("{}", format_args!("Battery: {0:.2}%", batter_percentage))`

Urhengulas, 2023-03-13:
Now I understand. We opted for the adapter types (`Display2Format` and `Debug2Format`) because users won't accidentally use the inefficient formatting but need to opt into it explicitly.

elpiel, 2023-03-13:
Ok, got it, thanks!

Qix-, 2023-10-19:
Is there a way to do this, though? I'm trying to write panic messages for when `core` panicks and it doesn't get picked up by defmt automatically, and `payload()` is a black box of information that sometimes cannot be downref'd to a `&'static str`.