knurling-rs/defmt #749: Make bumping the wire format an API breaking change

therealfrauholle, 2023-04-05:
Hey @Urhengulas after the latest release of the changes in https://github.com/knurling-rs/defmt/pull/747 I am still having problems aligning versions in my setup :sweat:

It appears to me that we missed bumping the dependency `defmt-macros` of `defmt` in https://github.com/knurling-rs/defmt/commit/37892413e3a64797bdc213d6a644754ac8bb2eef and also earlier in https://github.com/knurling-rs/defmt/commit/1cbbb334bf508d13c1d537623fe2a4165dfc5d65, it should now depend on `defmt-macros-0.3.4`.

Following this I tried to wrap my head around how the wire format plays together with the SemVer guarantees, and I think that bumping the wire format is an API breaking change.

This is related to the changes introduced as of https://github.com/knurling-rs/defmt/pull/540, which solved https://github.com/knurling-rs/defmt/issues/287.

Dirbaio, 2023-04-08:
The original rationale for decoupling the wire format version from the defmt semver version was that doing a defmt semver-major bump is a giant pain. You can't mix defmt versions in a binary (#426), and this seems fundamentally unfixable with the defmt design. So upgrading defmt requires ALL libraries using it to update. And in turn, that update requires a major bump of the library, because it'll break binaries that are still using the older version.

This happened in the defmt 0.2 -> 0.3 upgrade. It was a giant pain, it took MONTHS.

Decoupling the versions allows defmt to evolve while avoiding regularly inflicting this pain into the ecosystem. I really don't think we should walk back on this decision.

However, I remember discussing making defmt-decoder support multiple wire format versions, which it seems it never happened. Currently old defmt-decoder/probe-run versions support only wire `3`, newer ones support only `4`, so you have to keep switching between them if some projects have been updated and some other haven't. 

If the new versions added support for `4` instead of dropping support for `3` the experience would be much better: you upgrade defmt, you get an error from probe-run saying "please update probe-run", and then everything would be working again.

Dirbaio, 2023-04-12:
It'd be nicer to have a single defmt-decoder version support multiple wire format versions. If it's one-to-one bins ca nhack it by depending on multiple defmt-decoder versions at once, yes, but that work needs to be repeated for each project that uses defmt-decoder, and needs manual work to add new versions, vs just `cargo update`ing to a newer defmt-decoder.

datdenkikniet, 2023-04-12:
Requiring an update to projects using `defmt-decoder` sounds far less problematic, IMO. Crates actually using that are few and far between, and being able to support multiple wire versions so that actual users can safely `cargo install <decoder thing>` without having to worry that you can't decode logs from projects locked to older versions of `defmt` anymore without `cargo install`-ing all the time.