knurling-rs/defmt #763: defmt impls for time

burrbull, 2023-06-27:
It makes sense to have implementations for [`time` crate](https://docs.rs/time/0.3.22/time/) primitives.

jonathanpallant, 2023-06-27:
Maybe, but there's an almost unlimited number of third-party types we could special case.

Also, most embedded projects seem to use https://docs.rs/chrono/0.4.26/chrono/ and not time.

Urhengulas, 2023-07-07:
I agree that there is a potentially unbound number. But I also think that either us or them should implement it, so users can use these types easily with `defmt`. And since both `time` and `chrono` are bigger than `defmt` I guess it makes sense to do it on our side.

Long story short, I'd be okay with accepting a PR to add `Format` implements for the time crate, hidden behind a feature gate. Same for chrono.

burrbull, 2025-01-08:
I still think this should be done on `defmt` side. `chrono` and `time` are common used crates implementing basic time `types`.
Where `defmt` is "embedded only" logging framework.

jonathanpallant, 2025-01-09:
Yeah, but if chrono or time make breaking change bumps, we have to implement the traits for all common versions, which gets messy.

We're just about to commit to never doing a major version bump, so it's cleaner for them to implement our trait. And DJC said he would once we hit 1.0.

Urhengulas, 2025-01-09:
Also if we implement it on the `chrono` or `time` side we have access to private fields. I don't know if this is necessary in this case but it makes sense in the general case.