knurling-rs/defmt #770: Mention the need for Encoder in certain cases in documentation

Qix-, 2023-08-06:
Trying to implement `defmt` for a custom board/serial line was a little bit of a headache just reading through the documentation. Maybe some improvements to the docs can be made.

- At first, it wasn't clear that `defmt` was sending indices of strings to a host program to decode. I realize it's trivial now given the "deferred" part of "defmt" but just reading the getting started docs didn't really make that clear. It definitely wasn't clear you need a host program to decode them for you.
- After getting `defmt` integrated and sending bytes over the serial line, installing `defmt-print` and trying to use it, I spent a good hour tinkering with my system's serial config thinking the bytes weren't coming in correctly or something, because all of the frames were malformed. I looked at the encoding docs and thought `raw` meant "no fancy encoding".
- Nowhere in the `global_logger` nor the "Encoding" page does it mention that in order to send frames over the wire, you must use an `Encoder` to get them working.

I'm sure to people who use `defmt` often this is all trivial but as a complete newcomer to defmt (having previously stored all logging strings directly in the firmware itself), it wasn't obvious what was happening or how this was supposed to work, especially trying to implement my own wire debugging.

Perhaps some of the docs could mention these sorts of cases?

Urhengulas, 2023-08-07:
Hi @Qix- 👋🏾 

Thank you for your message! Making our documentation more beginner-friendly is definitely something we have to work on.

Would you be interested in proposing changes to the docs to make it more clear? Since you just went through the process you probably know quite well what would have helped you. If yes, please just send a PR. If not we can totally also work on it.

🥂

Qix-, 2023-08-07:
Sure thing! I'll try to get some time sometime this week to do it :) Thanks for the response.