knurling-rs/defmt #772: Cannot build program: linking with `rust-lld` failed

adrien3d, 2023-08-12:
Describe the bug
While trying to `cargo build`  this [project](https://github.com/adrien3d/rs-nrf-fw/tree/dev) on my Mac OS 13.5 M1, with rustc 1.71.1 (eb26296b5 2023-08-03) I have the following error:
```
Compiling rs-nrf-fw v0.1.0 (/Users/adrien/Dev/rust/rs-nrf-fw)
error: linking with `rust-lld` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/Users/adrien/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/bin:/opt/homebrew/sbin:/Users/adrien/.pyenv/shims:/Users/adrien/.pyenv/bin:/Users/adrien/.nvm/versions/node/v14.17.4/bin:/usr/local/sbin:/opt/homebrew/bin:/opt/homebrew/Cellar:/opt/homebrew/opt/make/libexec/gnubin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/gcc-arm-none-eabi-8-2019-q3-update/bin:/usr/local/nRF-Command-Line-Tools_10_13_0/mergehex:/usr/local/nRF-Command-Line-Tools_10_13_0/nrfjprog:/Users/adrien/pebble-dev/pebble-sdk-4.5-mac/bin:/usr/local/go/bin:/usr/local/mysql/bin:/opt/local/bin:/opt/local/sbin:/Users/adrien/Library/Android/sdk/tools:/Users/adrien/Library/Android/sdk/platform-tools:/Applications/Android Studio.app/Contents/gradle/gradle-4.1/bin/gradle:/Users/adrien/Dev/esp/xtensa-esp32-elf/bin:/Users/adrien/.cargo/bin:/usr/local/flutter/bin:/Users/adrien/.ebcli-virtual-env/executables:/Users/adrien/Dev/go/bin" VSLANG="1033" "rust-lld" "-flavor" "gnu" "/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/rustcXmbOaA/symbols.o" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/rs_nrf_fw-fbcd15a6ed501325.1tdc1q81f6y2xo16.rcgu.o" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/rs_nrf_fw-fbcd15a6ed501325.2pge09an1oysmzhd.rcgu.o" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/rs_nrf_fw-fbcd15a6ed501325.2vseympns95nevyu.rcgu.o" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/rs_nrf_fw-fbcd15a6ed501325.2x19p0pqt1niwuhs.rcgu.o" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/rs_nrf_fw-fbcd15a6ed501325.2x1o9uruh7b0ze5i.rcgu.o" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/rs_nrf_fw-fbcd15a6ed501325.32rf74fujkm6e9zf.rcgu.o" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/rs_nrf_fw-fbcd15a6ed501325.39krz5fn1dv4h0o7.rcgu.o" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/rs_nrf_fw-fbcd15a6ed501325.3mmigk47elnxh5pi.rcgu.o" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/rs_nrf_fw-fbcd15a6ed501325.3sedmhuoifsq2cun.rcgu.o" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/rs_nrf_fw-fbcd15a6ed501325.4plqnnjazbtbxxkw.rcgu.o" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/rs_nrf_fw-fbcd15a6ed501325.58srl0oyixd3a6ar.rcgu.o" "--as-needed" "-L" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps" "-L" "/Users/adrien/Dev/rust/rs-nrf-fw/target/debug/deps" "-L" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/build/cortex-m-965a8d5f596a4591/out" "-L" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/build/cortex-m-rt-3e08963052a67e28/out" "-L" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/build/defmt-1a56c5f4f8364f06/out" "-L" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/build/nrf52840-hal-6a04172d30d62ff5/out" "-L" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/build/nrf52840-pac-9b8ac9de771f2aac/out" "-L" "/Users/adrien/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/thumbv7em-none-eabihf/lib" "-Bstatic" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libnrf52840_hal-5c632d095bd9de47.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libnrf_hal_common-48df6db66e91692f.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libnrf_usbd-9df400a7ae71d633.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libusb_device-43398db52bcc1820.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libfixed-3a607f32876f7066.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libtypenum-e356c3e13b08d28b.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libbytemuck-f87704411a8ead67.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libhalf-60718bfc9ba4bd1d.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libaz-c96805a47fb398bc.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/librand_core-27e04ce0c49dca92.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libcfg_if-fffc2b18dcf0c77f.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libembedded_storage-7ea7f207a65f71c1.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libembedded_dma-500152b8fe0cd76e.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libstable_deref_trait-bd08b5e2c0781aa8.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libcast-95deeee22732f9ea.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libnrf52840_pac-b3596c856aae47c9.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libcortex_m_rt-2a27b9c1ff1387de.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libcortex_m-d1bf049e0fddb173.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libvolatile_register-e0f92e63c1c8882b.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libvcell-9fdf4fe90dc9191e.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libbare_metal-569bcb5fa6d0fe0f.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libdefmt_rtt-87c3f835ab1d189b.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libcritical_section-6a5d9c232f127ee3.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libdefmt-868172566efdc129.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libbitflags-75059a47b30b4ede.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libembedded_hal-95a0b41a4a40337a.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libvoid-67193c680d6b2ada.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libnb-381a2df034b2bc3a.rlib" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libnb-4b265eef857f001d.rlib" "/Users/adrien/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/thumbv7em-none-eabihf/lib/librustc_std_workspace_core-5828e8d35e98b226.rlib" "/Users/adrien/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/thumbv7em-none-eabihf/lib/libcore-0a4ed4069afedd7f.rlib" "/Users/adrien/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/thumbv7em-none-eabihf/lib/libcompiler_builtins-1cc8d288f9da4605.rlib" "-Bdynamic" "--eh-frame-hdr" "-z" "noexecstack" "-L" "/Users/adrien/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/thumbv7em-none-eabihf/lib" "-o" "/Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/rs_nrf_fw-fbcd15a6ed501325" "--gc-sections" "-Tlink.x" "-Tdefmt.x"
  = note: rust-lld: warning: section type mismatch for .uninit.defmt-rtt.BUFFER
          >>> /Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libdefmt_rtt-87c3f835ab1d189b.rlib(defmt_rtt-87c3f835ab1d189b.defmt_rtt.e7f7a56701267076-cgu.8.rcgu.o):(.uninit.defmt-rtt.BUFFER): SHT_PROGBITS
          >>> output section .uninit: SHT_NOBITS
          
          rust-lld: warning: section type mismatch for .got
          >>> <internal>:(.got): SHT_PROGBITS
          >>> output section .got: SHT_NOBITS
          
          rust-lld: warning: section type mismatch for .got.plt
          >>> <internal>:(.got.plt): SHT_PROGBITS
          >>> output section .got: SHT_NOBITS
          
          rust-lld: warning: section type mismatch for .got
          >>> <internal>:(.got): SHT_PROGBITS
          >>> output section .got: SHT_NOBITS
          
          rust-lld: error: undefined symbol: _critical_section_1_0_acquire
[... cut ...]
          >>> referenced by lib.rs:200 (/Users/adrien/.cargo/registry/src/index.crates.io-6f17d22bba15001f/critical-section-1.1.2/src/lib.rs:200)
          >>>               defmt_rtt-87c3f835ab1d189b.defmt_rtt.e7f7a56701267076-cgu.8.rcgu.o:(_$LT$defmt_rtt..Logger$u20$as$u20$defmt..traits..Logger$GT$::release::hec59024889505873) in archive /Users/adrien/Dev/rust/rs-nrf-fw/target/thumbv7em-none-eabihf/debug/deps/libdefmt_rtt-87c3f835ab1d189b.rlib
          

error: could not compile `rs-nrf-fw` (bin "rs-nrf-fw") due to previous error
```

Urhengulas, 2023-08-23:
Can you please try what happens if you execute it with `probe-rs run` instead of `probe-run`?

You can install the `probe-rs` cli with:
```console
$ cargo install probe-rs --features cli
```

And then change your `.cargo/config.toml` like this:
```diff
[target.'cfg(all(target_arch = "arm", target_os = "none"))']
-runner = "probe-run --chip nRF52840_xxAA"
+runner = "probe-rs run --chip nRF52840_xxAA"
```

Urhengulas, 2023-08-23:
What also seems off is the fact that your `Running ...` line does not mention `probe-run`. It should look like following:

```console
$ cargo rb hello
    Finished dev [optimized + debuginfo] target(s) in 0.02s
     Running `probe-run --chip nRF52840_xxAA target/thumbv7em-none-eabihf/debug/hello`
```