knurling-rs/defmt #788: book: Acquisition outdated

chrysn, 2023-10-12:
At https://defmt.ferrous-systems.com/acq-rel, the book talks about `if let Some(logger) = Logger::acquire() {`. However, the latest API documentation (and source code) have `acquire()` return the unit value.

I don't have any fix to offer for the book, because I'd rely on the book to understand how it's supposed to be used :-)

(In particular, I'd love to make a note that I've dropped a log line and make the writer stop writing, because for my logger I neither want to create a critical section nor multiple channels -- and given I'm working with a multithreaded OS I can't just keep ignoring data sent from the later caller until its release is called, for the other thread may continue -- but that's off topic for here, as this issue is just about the inconsistency between book and docs).

amanjeev, 2023-10-19:
I believe this is an error on the book part. I see the API changed quite some time ago and perhaps the example was not updated.  Thank you for reporting this and apologies it is confusing.  I will add this to our triage.

Urhengulas, 2024-12-02:
Closed in favor of https://github.com/knurling-rs/defmt/issues/906.