knurling-rs/defmt #790: Migration to probe-rs

andresv, 2023-10-16:
I followed migration guide from https://ferrous-systems.com/blog/probe-run-deprecation/.
I am using latest `defmt`, `probe-rs`, `defmt-rtt`.
Using after migration I see this (using STM32F411CEUx ):

```
43.380859 INFO  usb rx[5, 5]: [04, 04, 03, FF, 00]
└─ <mod path> @ └─ <invalid location: defmt frame-index: 27>:0
43.381103 INFO  in6 X
└─ <mod path> @ └─ <invalid location: defmt frame-index: 20>:0
43.394744 INFO  usb rx[5, 5]: [03, 04, 03, 01, 00]
└─ <mod path> @ └─ <invalid location: defmt frame-index: 27>:0
```

BriocheBerlin, 2023-10-17:
Hey, unfortunately I don't have a STM32F411CEUx at hand (also this is not a lot of context).
What is the (long form of the) command you are running? Are you by any chance running a `--release`? Wondering if [this](https://github.com/embassy-rs/embassy/wiki/Running-the-Examples#strange-output-on-the-console) might apply.

andresv, 2023-10-17:
That's it! I messed up Cargo workspace and therefore `debug = 2` wasn't used.

closed 2023-10-17 by andresv