knurling-rs/defmt #810: Documentation for logger formatting

Sympatron, 2024-02-07:
As far as I can see logger formatting is not really documented on .
You can kind of see how it works from several PRs: https://github.com/knurling-rs/defmt/pull/778#issue-1921027129, https://github.com/knurling-rs/defmt/pull/769#issue-1831997750, https://github.com/knurling-rs/defmt/pull/765#issue-1806095785. But this is hard to find and not very comprehensive.

Could somebody add documentation for this feature or point me to it if I just didn't find it?

andresovela, 2024-02-09:
I added a [page](https://github.com/knurling-rs/defmt/blob/main/book/src/custom-log-output.md) to the book to document that feature, but I guess the book hasn't been updated with that change yet.

Sympatron, 2024-02-13:
Great! Can somebody please publish the current version of the book to https://defmt.ferrous-systems.com then so people can find it more easily? Or are there any blockers?

closed 2024-03-05 by Urhengulas