knurling-rs/defmt #814: Licenses are missing copyright information

arctic-alpaca, 2024-02-27:
Hi,

both the Apache 2 and the MIT license are missing the copyright information.
For the Apache 2 license, the copyright placeholder is not filled in: https://github.com/knurling-rs/defmt/blob/main/LICENSE-APACHE#L189
For the MIT license, the copyright part at the top (`Copyright <YEAR> <COPYRIGHT HOLDER>`, see [here](https://opensource.org/license/mit)) is missing.

jonathanpallant, 2024-02-27:
The copyright line in that part of the Apache-2.0 licence is a template and is not supposed to be modified in the licence file itself. If you changed it, then you have a) modified the licence, and b) are telling anyone who reads your modified copy of the licence that they must put your copyright in their files when they wish to release it under your modified Apache-2.0 licence.

The copyright line is missing in the MIT file and we should fix that. I note it is also missing from the copy at GitHub.com/rust-lang so maybe we copied their file.

arctic-alpaca, 2024-02-27:
You are correct, the Apache-2.0 license file should not be modified, my bad. Without any form of copyright notice, I'm unsure how the copyright is assigned when using this project under the Apache-2.0 license.

The Rust repository has a [COPYRIGHT file](https://github.com/rust-lang/rust/blob/master/COPYRIGHT).

Urhengulas, 2024-12-02:
We do have https://github.com/knurling-rs/defmt/blob/f5315e098129c372d32ed6fc0fe309d6302488ee/LICENSE-MIT#L1 is this not enough?

jonathanpallant, 2024-12-02:
I added that five days ago :)

Urhengulas, 2024-12-02:
Perfect

closed 2024-12-02 by Urhengulas