knurling-rs/defmt #817: Use UnsafeCell instead of static-mut

jonathanpallant, 2024-03-07:
The code at https://github.com/knurling-rs/defmt/blob/4db33ca5df8d7a7b5d9036c49b1ea535cc3986c0/firmware/defmt-rtt/src/lib.rs#L140 uses a `static mut`. It goes to some trouble to ensure that the `handle()` function cannot be called twice, but perhaps we can use the type system to guarantee that is the case. Or if not, maybe we should use an `UnsafeCell`.

Also, the buffer is placed in an `uninit` section but we don't use the `MaybeUninit` type for the buffer. If there's a reason, we should perhaps explain why.

Urhengulas, 2024-03-07:
@japaric Can you shed some light on if there is a reason it should be implemented like it is right now?

evading, 2024-04-19:
> Also, the buffer is placed in an `uninit` section but we don't use the `MaybeUninit` type for the buffer. If there's a reason, we should perhaps explain why.

I was just about to file an issue about that. In the commit that introduced the `uninit` section in `cortex-m-rt` it says that `MaybeUninit` is the only correct way to use that section [1].

[1]: https://github.com/rust-embedded/cortex-m/commit/94fbbe01189f97aefa8d6a97cc7cd73e5c6f6ec9