knurling-rs/defmt #818: defmt_rtt::flush implementation does not work with `probe-rs`

Urhengulas, 2024-03-07:
I just realized that `probe-rs` does not set RTT to blocking, at least that is how I understand the documentation ([link](https://github.com/probe-rs/probe-rs/blob/e42b0badd7c2c90e41ceea1b8ca8e69731249fc4/probe-rs/src/rtt.rs#L3-L6)). While this is generally good, this means the `defmt-rtt` cannot detect that `probe-rs` is connected and therefore the `flush` implementation does not work anymore.

https://github.com/knurling-rs/defmt/blob/ae2d0dff82014e334d6c2480cb59b95682408ff5/firmware/defmt-rtt/src/channel.rs#L103-L106

Any ideas how to detect it otherwise?

bugadani, 2024-03-12:
I think it's a clever enough idea and we can configure channels to blocking in probe-rs if we can prove they are defmt-related up channels.

Urhengulas, 2024-03-20:
> I think it's a clever enough idea and we can configure channels to blocking in probe-rs if we can prove they are defmt-related up channels.

That's a good idea