knurling-rs/defmt #820: [defmt-test] Not possible to show location of test functions in stdout logger

mhatzl, 2024-03-12:
## Problem

With the changes introduced in PR #765 it is not possible to show the location of test functions with the stdout logger.
Only `(<test nr>/<nr of tests>) running <fn name> ...` will be printed.

## Cause

Since PR #765, prints are not formatted and only the print message is output to stdout.
See commit 431fb80 for the changes that caused this behavior.

According to PR #765, the decision to not format print statements was intentional.
The problem is that defmt-test uses println! to log the start of a test function (see https://github.com/knurling-rs/defmt/blob/main/firmware/defmt-test/macros/src/lib.rs#L410).
This log is not really useful without location information in my opinion.

## Expected Solution

Display location information for test functions as shown in the README of defmt-test.

```
$ cargo test --lib
(..)
(1/1) running `it_works`...
└─ app::unit_tests::__defmt_test_entry @ src/lib.rs:33
all tests passed!
└─ app::unit_tests::__defmt_test_entry @ src/lib.rs:28
(..)
(HOST) INFO  device halted without error
```

## Possible Fix

The easiest fix might be to use the `info` level instead of `println` to log the start of a test function.

Urhengulas, 2024-03-20:
I am not sure if we should add the path. The normal `cargo test` runner does not either.

mhatzl, 2024-03-20:
This is also fine for me.
In that case the screenshots/samples should be updated, because I thought something was wrong in my setup.