knurling-rs/defmt #823: cargo-embed produces broken output

FlareFlo, 2024-03-25:
When running my [learning project](https://github.com/FlareFlo/stm32-rtic/tree/defmt-rtt-issue)` using ` using cargo-embed, the defmt output appears to be complete garbage.
Running with `cargo-embed --release`:
[image]
When running with probe-rs directly after the output appears to be correct:
[image]
I hope my project is minimal enough to help with reproducing the issue. The hardware used is an STM32F411CEb with the STLink V2.0.
Cargo-embed is on version 0.13.0
probe-rs is on version 0.23.0

Urhengulas, 2024-03-26:
Can you please report this at probe-rs?

FlareFlo, 2024-03-26:
Done

closed 2024-03-26 by FlareFlo