knurling-rs/defmt #826: View defmt_rtt output in J-Link viewer

pnchatterji, 2024-04-04:
Hello,
I am using defmt_rtt in an Embassy application. In order to view output messages using probe_rs, I was forced to change the jlink usb driver. I had previously attempted to view the output on J-Link viewer, but I was getting junk, even after setting encoding to raw. However, now my J-link viewer is unusable. Probably I would need to reinstall it. Is there any way to see the  defmt_rtt output in j-link viewer, thus avoiding having to replace the USB driver? I need the j-link viewer for non-rust embedded applications, so I can't keep installing and reinstalling the USB drivers.

thanks
regards
pnc

jonathanpallant, 2024-04-17:
I think you would need to talk to SEGGER about adding support for the defmt protocol to their J-Link tool, either directly or by piping the RTT byte stream through the `defmt-print` binary.

kwalker-plxs, 2026-03-11:
Hello @jonathanpallant and @pnchatterji, 

Did one of you end up contacting SEGGER about this?