knurling-rs/defmt #833: defmt 0.3.7 breaks build for older versions of rust

astapleton, 2024-05-13:
It seems the change to deprecate the defmt `ip_in_core` feature #805 breaks compilation of defmt using older versions of rust  that do not have the feature enabled (I assume <1.77, but I'm seeing the failure on 1.69) and it doesn't seem like there's an option to opt out of compiling that support in? I see defmt makes no guarantees around what version it supports, so is this just a case where we should pin the version of defmt to an older version?

Urhengulas, 2024-05-14:
Can you please share the error message?

astapleton, 2024-05-14:
Sorry! Should have included that to begin with. Updated the top level comment with the error message

thvdveld, 2024-05-16:
This seems to work, but I had to remove the heapless defmt-03 feature in smoltcp to make it work.

```
error: failed to select a version for `defmt`.
    ... required by package `heapless v0.8.0`
    ... which satisfies dependency `heapless = "^0.8"` of package `smoltcp v0.11.0 (/home/thvdveld/source/vub/smoltcp)`
versions that meet the requirements `>=0.2.0, <0.4` are: 0.3.7, 0.3.6, 0.3.5, 0.3.4, 0.3.2, 0.3.0, 0.2.3, 0.2.2, 0.2.1, 0.2.0

the package `defmt` links to the native library `defmt`, but it conflicts with a previous package which links to `defmt` as well:
package `defmt v0.3.7 (https://github.com/knurling-rs/defmt?branch=support-pre-1.77#e9290117)`
    ... which satisfies git dependency `defmt` of package `smoltcp v0.11.0 (/home/thvdveld/source/vub/smoltcp)`
Only one package in the dependency graph may specify the same links value. This helps ensure that only one copy of a native library is linked in the final binary. Try to adjust your dependencies so that only one package uses the `links = "defmt"` value. For more information, see https://doc.rust-lang.org/cargo/reference/resolver.html#links.

failed to select a version for `defmt` which could resolve this conflict
```

I think this is because the `support-pre-1.77` branch still uses 0.3.7 as a version.

arctic-alpaca, 2024-05-17:
I think adding `ip_in_core` to the default features is a breaking change, but the [reference](https://doc.rust-lang.org/cargo/reference/semver.html#cargo-feature-add) has no clear answer for this. Since this will break compilation for every project using defmt (without `default-features = false`) with older Rust versions, I would consider this breaking.

I'm seeing the same issue as @thvdveld with other dependencies depending on defmt.

Urhengulas, 2024-05-17:
@thvdveld 
> I think this is because the `support-pre-1.77` branch still uses 0.3.7 as a version.

@arctic-alpaca 
> I'm seeing the same issue as @thvdveld with other dependencies depending on defmt.

Yes, the problem is that there can only be one version of `defmt` in your dependency tree and heapless depends on the published version while you use the git version. This will go away as soon as it is published.

Urhengulas, 2024-05-17:
> I think adding `ip_in_core` to the default features is a breaking change, but the [reference](https://doc.rust-lang.org/cargo/reference/semver.html#cargo-feature-add) has no clear answer for this. [...]

I am fine with making it opt-in instead.

Urhengulas, 2024-05-17:
@arctic-alpaca @thvdveld I changed the PR to opt-in. I think it should work for you, but maybe you wanna try again?

thvdveld, 2024-05-17:
Checked it and it works. Thank you!

arctic-alpaca, 2024-05-17:
Also works on my end, thank you!

Urhengulas, 2024-05-17:
Nice. Gonna merge, release `defmt 0.3.8` and yank `0.3.7`.

closed 2024-05-17 by Urhengulas

Urhengulas, 2024-05-17:
defmt 0.3.8 was just released

astapleton, 2024-05-17:
Also works here. Thanks!

PR #840 (2024-05-17): Support pre 1.77
This PR makes defmt compatible with versions of rustc smaller than `1.77`. It achieves this by allowing to exclude `defmt::Format` implementations for types which got stabilized under the `ip_in_core` feature. Note that those types are included by default, but can be excluded by disabling the default cargo features.

Fixes #833