knurling-rs/defmt #834: defmt 0.3.7 breaks embedded-hal with the `defmt-03` feature flag

guineawheek, 2024-05-13:
Builds now fail like this:

```rust

error[E0433]: failed to resolve: could not find `defmt` in the list of imported crates
   --> /home/guinea/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-hal-1.0.0/src/i2c.rs:192:41
    |
192 | #[cfg_attr(feature = "defmt-03", derive(defmt::Format))]
    |                                         ^^^^^^^^^^^^^ could not find `defmt` in the list of imported crates
    |
    = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)
help: consider importing this crate through its public re-export
    |
164 + use crate::defmt;
    |

error[E0433]: failed to resolve: could not find `defmt` in the list of imported crates
   --> /home/guinea/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-hal-1.0.0/src/i2c.rs:192:41
    |
192 | #[cfg_attr(feature = "defmt-03", derive(defmt::Format))]
    |                                         ^^^^^^^^^^^^^ could not find `defmt` in the list of imported crates
    |
    = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0433]: failed to resolve: could not find `defmt` in the list of imported crates
   --> /home/guinea/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-hal-1.0.0/src/i2c.rs:304:41
    |
304 | #[cfg_attr(feature = "defmt-03", derive(defmt::Format))]
    |                                         ^^^^^^^^^^^^^ could not find `defmt` in the list of imported crates
    |
    = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)
help: consider importing this crate through its public re-export
    |
164 + use crate::defmt;
    |

error[E0433]: failed to resolve: could not find `defmt` in the list of imported crates
   --> /home/guinea/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-hal-1.0.0/src/i2c.rs:304:41
    |
304 | #[cfg_attr(feature = "defmt-03", derive(defmt::Format))]
    |                                         ^^^^^^^^^^^^^ could not find `defmt` in the list of imported crates
    |
    = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0433]: failed to resolve: could not find `defmt` in the list of imported crates
   --> /home/guinea/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-hal-1.0.0/src/spi.rs:201:41
    |
201 | #[cfg_attr(feature = "defmt-03", derive(defmt::Format))]
    |                                         ^^^^^^^^^^^^^ could not find `defmt` in the list of imported crates
    |
    = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)
help: consider importing this crate through its public re-export
    |
174 + use crate::defmt;
    |

error[E0433]: failed to resolve: could not find `defmt` in the list of imported crates
   --> /home/guinea/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-hal-1.0.0/src/spi.rs:201:41
    |
201 | #[cfg_attr(feature = "defmt-03", derive(defmt::Format))]
    |                                         ^^^^^^^^^^^^^ could not find `defmt` in the list of imported crates
    |
    = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0433]: failed to resolve: could not find `defmt` in the list of imported crates
   --> /home/guinea/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-hal-1.0.0/src/spi.rs:319:41
    |
319 | #[cfg_attr(feature = "defmt-03", derive(defmt::Format))]
    |                                         ^^^^^^^^^^^^^ could not find `defmt` in the list of imported crates
    |
    = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)
help: consider importing this crate through its public re-export
    |
174 + use crate::defmt;
    |

error[E0433]: failed to resolve: could not find `defmt` in the list of imported crates
   --> /home/guinea/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-hal-1.0.0/src/spi.rs:319:41
    |
319 | #[cfg_attr(feature = "defmt-03", derive(defmt::Format))]
    |                                         ^^^^^^^^^^^^^ could not find `defmt` in the list of imported crates
    |
    = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)
```

`cargo -V` is `cargo 1.80.0-nightly (4de0094ac 2024-05-09)`

diondokter, 2024-05-13:
This seems to happen from the update of the macros crate specifically

diondokter, 2024-05-13:
Ok, it specifically has to do with the renaming of the dependency in eh. They renamed the crate `defmt-03` and then have a `use defmt_03 as defmt;` in the lib.rs so they can manage versions

diondokter, 2024-05-13:
I think I found it: https://github.com/knurling-rs/defmt/pull/800/files#diff-2a03026349ac5c67a33820bc72e0da8740c03b6b6dbdc9926cdac9ce0c69c2d2R53

It changes from `defmt` to `::defmt`. Not too sure about what the rules are with this though.

PR #835 (2024-05-14): Use the more naive crate name to support renaming
Fixes #834 

`::defmt` points to the crate called `defmt`. This is nice to use because it avoids bad renames. One example I saw was `use mini_serde as serde`.

However, people have started renaming the crate in their own crates. Notably, embedded-hal does this which renames it `defmt-03`.
With the inclusion of defmt in these kinds of foundational crates, they will want to be able to support multiple defmt versions at the same time and they kinda need the renaming. (I don't see another way to do it)

Previously before #800 this all worked, but t [... cut]
closed 2024-05-14 by Urhengulas